espressif/esp-qcloud #2: 这个demo怎么都通不过

zilanghe, 2018-12-18:
ERR|1970-01-01 00:00:13|HAL_TLS_mbedtls.c|HAL_TLS_Connect(253): mbedtls_ssl_hand
shake failed returned 0x6800
DBG|1970-01-01 00:00:13|HAL_TLS_mbedtls.c|HAL_TLS_Disconnect(277): handle is NUL
L
ERR|1970-01-01 00:00:13|mqtt_client.c|IOT_MQTT_Construct(90): mqtt connect with
id:  failed: -1001
E (13548) esp32-qcloud-mqtt-demo: Cloud Device Construct Failed

idf版本为3.0.7

xiewenxiang, 2018-12-28:
@zilanghe  看你的log， 应该是key 设置的不对吧，烧录的key 正确吗？

zilanghe, 2018-12-30:
@xiewenxiang

正确的。烧录同样的固件，多次运行，有时subscribe能成功，多数时候不能；即使subscribe成功，也会返回-111的错误代码。

Eavo, 2019-01-03:
谢谢您的建议，当前我们计划重构 & 升级 Qcloud SDK，之后将基于新的 SDK 测试这个问题，届时也欢迎您和我们一起测试。

jsonsiyuan, 2019-12-13:
需要再sdk中开启一些TLS的密钥协商方式

jsonsiyuan, 2020-01-07:
TLS握手失败 使用的是密钥认证，需要 Enable PSK based ciphersuite modes

closed 2020-03-26 by xiewenxiang