espressif/esp-qcloud #5: SDK的问题

Jiangwei0802, 2020-01-06:
1.编译不通过：提示缺少一些DTLS相关 API定义
[image]

2.文档与实际SDK里面设置不符合
这个demo的readme说明有勘误，这是一个没有使用证书认证的demo，而是使用了设备密钥认证的demo。代码中没有开始AUTH_MODE_CERT

3.TLS握手失败

4.腾讯云的几种认证方式目前是否   证书认证、密钥认证、动态认证都支持？

jsonsiyuan, 2020-01-07:
对于TLS握手失败 需要开启 密钥协商中的 Enable PSK based ciphersuite modes

songjiaxiu-espressif, 2020-01-10:
@Jiangwei0802 
     您好。
1. 编译不通过：提示缺少一些DTLS相关 API定义
                  -----------------------------------------------
要在 `make menuconfig` -> `Component config` -> `mbedTLS` -> `support DTLS protocol` 选中，如下图所示：

[image]

2. 文档与实际SDK里面设置不符合
               ---------------------------------------------------
文档目前还没有更新，这个 `demo`, 对于证书方式和设备认证方式都是支持的，其中 `AUTH_MODE_CERT` 定义分别针对 `cmake` 和 `make` 的编译方式，在目录下 `CMakLists.txt`,`component.mk` 文件中选择是否定义.

3.TLS握手失败
            ------------------------------------------------------
同样在 `make menuconfig` 中 Component config > mbedTLS > TLS Key Exchange Methods 中设置 `Enable pre-shared-key ciphersuites` 选中，如下图所示：

[image]

4. 目前支持证书认证、密钥认证

针对1和3问题，`demo` 下的 `sdkconfig.defaults` 中做了默认配置。

wenkumk, 2020-02-20:
> @Jiangwei0802
> 您好。
> 
> 1. 编译不通过：提示缺少一些DTLS相关 API定义
>    -----------------------------------------------
>    要在 `make menuconfig` -> `Component config` -> `mbedTLS` -> `support DTLS protocol` 选中，如下图所示：
> 
> [image]
> 
> 1. 文档与实际SDK里面设置不符合
>    ---------------------------------------------------
>    文档目前还没有更新，这个 `demo`, 对于证书方式和设备认证方式都是支持的，其中 `AUTH_MODE_CERT` 定义分别针对 `cmake` 和 `make` 的编译方式，在目录下 `CMakLists.txt`,`component.mk` 文件中选择是否定义.
> 
> 3.TLS握手失败
> ------------------------------------------------------
> 同样在 `make menuconfig` 中 Component config > mbedTLS > TLS Key Exchange Methods 中设置 `Enable pre-shared-key ciphersuites` 选中，如下图所示：
> 
> [image]
> 
> 1. 目前支持证书认证、密钥认证
> 
> 针对1和3问题，`demo` 下的 `sdkconfig.defaults` 中做了默认配置。

[image]
编译后现在有这几个报错是什么愿意呢

xiewenxiang, 2020-03-26:
如果还有问题，可以重新打开这个issues

closed 2020-03-26 by xiewenxiang