espressif/esp-qcloud #6: 编译出以下问题

wenkumk, 2020-01-16:
E:/ESP8266/TOOLS/msys32/home/Administrator/esp/esp-qcloud/examples/mqtt_demo/build/esp-qcloud\libesp-qcloud.a(HAL_DTLS_mbedtls.o):(.literal._free_mebeddtls+0x0): undefined reference to `mbedtls_ssl_cookie_free'
E:/ESP8266/TOOLS/msys32/home/Administrator/esp/esp-qcloud/examples/mqtt_demo/build/esp-qcloud\libesp-qcloud.a(HAL_DTLS_mbedtls.o):(.literal.HAL_DTLS_Connect+0x10): undefined reference to `mbedtls_ssl_cookie_check'
E:/ESP8266/TOOLS/msys32/home/Administrator/esp/esp-qcloud/examples/mqtt_demo/build/esp-qcloud\libesp-qcloud.a(HAL_DTLS_mbedtls.o):(.literal.HAL_DTLS_Connect+0x14): undefined reference to `mbedtls_ssl_cookie_write'
E:/ESP8266/TOOLS/msys32/home/Administrator/esp/esp-qcloud/examples/mqtt_demo/build/esp-qcloud\libesp-qcloud.a(HAL_DTLS_mbedtls.o):(.literal.HAL_DTLS_Connect+0x3c): undefined reference to `mbedtls_ssl_cookie_setup'
E:/ESP8266/TOOLS/msys32/home/Administrator/esp/esp-qcloud/examples/mqtt_demo/build/esp-qcloud\libesp-qcloud.a(HAL_DTLS_mbedtls.o): In function `_free_mebeddtls':
E:/ESP8266/TOOLS/msys32/home/Administrator/esp/esp-qcloud/qcloud_iot_c_sdk/platform/HAL_DTLS_mbedtls.c:78: undefined reference to `mbedtls_ssl_cookie_free'
E:/ESP8266/TOOLS/msys32/home/Administrator/esp/esp-qcloud/examples/mqtt_demo/build/esp-qcloud\libesp-qcloud.a(HAL_DTLS_mbedtls.o): In function `HAL_DTLS_Connect':
E:/ESP8266/TOOLS/msys32/home/Administrator/esp/esp-qcloud/qcloud_iot_c_sdk/platform/HAL_DTLS_mbedtls.c:225: undefined reference to `mbedtls_ssl_cookie_setup'
E:/ESP8266/TOOLS/msys32/home/Administrator/esp/esp-qcloud/examples/mqtt_demo/build/esp-qcloud\libesp-qcloud.a(HAL_DTLS_mbedtls.o): In function `HAL_DTLS_Disconnect':
E:/ESP8266/TOOLS/msys32/home/Administrator/esp/esp-qcloud/qcloud_iot_c_sdk/platform/HAL_DTLS_mbedtls.c:308: undefined reference to `mbedtls_ssl_cookie_free'
collect2.exe: error: ld returned 1 exit status
make: *** [E:/ESP8266/TOOLS/msys32/home/Administrator/esp/ESP8266_RTOS_SDK/make/project.mk:406：/home/Administrator/esp/esp-qcloud/examples/mqtt_demo/build/esp-qcloud.elf] 错误 1

songjiaxiu-espressif, 2020-01-17:
这个问题详见: https://github.com/espressif/esp-qcloud/issues/5
先关掉这个issue了，有问题可再次提问或再次打开此 issue.

songjiaxiu-espressif, 2020-01-17:
closed #6

closed 2020-02-20 by wenkumk