espressif/esp-qcloud #8: 编译通过后，tcp连不上

wenkumk, 2020-02-21:
[image]
如图，crt，这些文件以替换，用mqttfx，可以连上，上板子就不行了

songjiaxiu-espressif, 2020-02-28:
请问是使用的板子是 `ESP8266`  还是 `ESP32` 呢？提供一下具体的 `idf` 的 `commit` 版本号，方便我们这边复现一下呢

wenkumk, 2020-02-28:
> 请问是使用的板子是 `ESP8266` 还是 `ESP32` 呢？提供一下具体的 `idf` 的 `commit` 版本号，方便我们这边复现一下呢

$ git rev-parse HEAD
d3517e35f4dfd587f6b965a173c6ef95d1653e84

$ git log --oneline --graph --decorate
*   d3517e35 (HEAD, tag: v3.3-rc1) Merge branch 'bugfix/support_GD25Q64_qio_enable_v3.3' into 'release/v3.3'
|\
| * 5a0e946a bugfix(flash): add support for GD25Q64 qio enable
* |   b22182f7 Merge branch 'feature/mbedtls_using_util_algorithm_v3.3' into 'release/v3.3'
|\ \
| * | ffaaf070 feat(mbedtls): mbedtls using SHA/AES of util component instead of its raw algorithm
| |/
* |   6222b1c1 Merge branch 'bugfix/fix_sha_data_overflow_v3.3' into 'release/v3.3'

songjiaxiu-espressif, 2020-02-28:
请在 `make menuconfig` 中使能 ` TLS Key Exchange Methods` 的 `Enable PSK based ciphersuite modes ` （如下图）,  试一下。如果还是不能正常工作的话，提供一下 `sdkconfig` 文件。
[image]

wenkumk, 2020-02-28:
INF|1970-01-01 00:00:05|qcloud_iot_device.c|iot_device_info_set(67): SDK_Ver: 3.1.2, Product_ID: UH5D9RXBVS, Device_Name: test
DBG|1970-01-01 00:00:05|HAL_TLS_mbedtls.c|HAL_TLS_Connect(209): Setting up the SSL/TLS structure...
DBG|1970-01-01 00:00:05|HAL_TLS_mbedtls.c|HAL_TLS_Connect(251): Performing the SSL/TLS handshake...
DBG|1970-01-01 00:00:05|HAL_TLS_mbedtls.c|HAL_TLS_Connect(252): Connecting to /UH5D9RXBVS.iotcloud.tencentdevices.com/8883...
ERR|1970-01-01 00:00:06|HAL_TLS_mbedtls.c|HAL_TLS_Connect(259): mbedtls_ssl_handshake failed returned 0x7780
DBG|1970-01-01 00:00:06|HAL_TLS_mbedtls.c|HAL_TLS_Disconnect(285): handle is NULL
ERR|1970-01-01 00:00:06|mqtt_client.c|IOT_MQTT_Construct(121): mqtt connect with id:  failed: -1001
INF|1970-01-01 00:00:06|mqtt_client.c|qcloud_iot_mqtt_fini(425): release mqtt client resources

[sdkconfig.zip](https://github.com/espressif/esp-qcloud/files/4266114/sdkconfig.zip)

songjiaxiu-espressif, 2020-02-28:
请问，所采取的加密方式是 `设备密钥认证` 还是 `证书认证` 的方式呢？
如果是前者，下图中的 `设备密钥` 要填写正确的内容，关于这一部分详见腾讯云：https://cloud.tencent.com/document/product/634/35274
[image]

wenkumk, 2020-02-28:
DBG|1970-01-01 00:00:05|HAL_TLS_mbedtls.c|HAL_TLS_Connect(209): Setting up the SSL/TLS structure...
DBG|1970-01-01 00:00:05|HAL_TLS_mbedtls.c|HAL_TLS_Connect(251): Performing the SSL/TLS handshake...
DBG|1970-01-01 00:00:05|HAL_TLS_mbedtls.c|HAL_TLS_Connect(252): Connecting to /497QN28C89.iotcloud.tencentdevices.com/8883.
..
ERR|1970-01-01 00:00:11|HAL_TLS_mbedtls.c|_mbedtls_tcp_connect(165): tcp connect failed returned 0x0052 errno: 0
DBG|1970-01-01 00:00:11|HAL_TLS_mbedtls.c|HAL_TLS_Disconnect(285): handle is NULL
ERR|1970-01-01 00:00:11|mqtt_client.c|IOT_MQTT_Construct(121): mqtt connect with id:  failed: -1001
INF|1970-01-01 00:00:11|mqtt_client.c|qcloud_iot_mqtt_fini(425): release mqtt client resources

重新录录入密匙认证证书，现在又回到这个BUG了

songjiaxiu-espressif, 2020-02-28:
如果是设备密钥认证的话，输入的设备密钥，并不是证书的。
我们支持 `设备密钥认证` 和 `证书认证` 两种方式，请问，所采取的加密方式是 `设备密钥认证` 还是 `证书认证` 的方式呢？

wenkumk, 2020-02-28:
INF|1970-01-01 00:00:05|qcloud_iot_device.c|iot_device_info_set(67): SDK_Ver: 3.1.2, Product_ID: XXXXXX, Device_Name: test
DBG|1970-01-01 00:00:05|HAL_TLS_mbedtls.c|HAL_TLS_Connect(209): Setting up the SSL/TLS structure...
DBG|1970-01-01 00:00:05|HAL_TLS_mbedtls.c|HAL_TLS_Connect(251): Performing the SSL/TLS handshake...
DBG|1970-01-01 00:00:05|HAL_TLS_mbedtls.c|HAL_TLS_Connect(252): Connecting to /XXXXXXXX.iotcloud.tencentdevices.com/8883...
INF|1970-01-01 00:00:05|HAL_TLS_mbedtls.c|HAL_TLS_Connect(274): connected with /XXXXXXX.iotcloud.tencentdevices.com/8883...
INF|1970-01-01 00:00:05|mqtt_client.c|IOT_MQTT_Construct(128): mqtt connect with id: XXXXX success
I (5973) esp32-qcloud-mqtt-demo: Cloud Device Construct Success
DBG|1970-01-01 00:00:05|mqtt_client_subscribe.c|qcloud_iot_mqtt_subscribe(138): topicName=XXXXXXXX/test/data|packet_id=XXXXX
I (6045) esp32-qcloud-mqtt-demo: subscribe success, packet-id=XXXXX
INF|1970-01-01 00:00:05|mqtt_client_connect.c|qcloud_iot_mqtt_disconnect(437): mqtt disconnect!
INF|1970-01-01 00:00:05|mqtt_client.c|IOT_MQTT_Destroy(189): mqtt release!

songjiaxiu-espressif, 2020-03-02:
您采用的认证方式是设备密钥认证，后台的 `log` 是正常的，`demo` 不支持 `smartlink` .
关于产品密钥详见：https://cloud.tencent.com/document/product/634/35273

xcguang, 2020-03-02:
官方有一套和 腾讯合作的 QCloud AT 固件，支持 smartconfig ，固件可以从 https://www.espressif.com/zh-hans/support/download/at 找到 `ESP8266 Qcloud IoT AT` 下载

xiewenxiang, 2020-03-26:
如果还有问题，可以重新打开这个issues

closed 2020-03-26 by xiewenxiang