espressif/esp-qcloud #10: 我增加了一个uart读取数据，初始化出了问题

jenking, 2021-02-17:
你好，我单独使用esp-qcloud的时候正常，准备增加uart1来读取传感器，在初始化的时候出了问题，提示如下:
I (742) esp_qcloud_log: log initialized successfully
E (755) app_main: The device has been restarted abnormally
I (758) uart: queue free spaces: 20
E (759) MB_PORT_COMMON: xMBMasterPortTimersInit(114): timer set value failure, timer_isr_register() returned (0x105).
E (769) MB_CONTROLLER_MASTER: mbc_serial_master_start(107): mb stack initialization failure, eMBInit() returns (0x3).
E (781) MB_CONTROLLER_MASTER: mbc_master_start(184): Master start failure, error=(0x103) (ESP_ERR_INVALID_STATE).
E (791) MASTER_TEST: master_init(298): mb controller start fail, returns(0x103).ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4002a52c
0x4002a52c: _esp_error_check_failed at D:/esp/esp-idf/components/esp_common/src/esp_err.c:41

file: "../main/app_main.c" line 318
func: app_main
expression: master_init()

abort() was called at PC 0x4002a52f on core 0
0x4002a52f: _esp_error_check_failed at D:/esp/esp-idf/components/esp_common/src/esp_err.c:42

Backtrace:0x4002c24b:0x3ffd6040 0x4002caed:0x3ffd6060 0x40033802:0x3ffd6080 0x4002a52f:0x3ffd60f0 0x4008bfc8:0x3ffd6110 0x4013877b:0x3ffd61f0 0x4002eb1d:0x3ffd6210
0x4002c24b: panic_abort at D:/esp/esp-idf/components/esp_system/panic.c:349

0x4002caed: esp_system_abort at D:/esp/esp-idf/components/esp_system/system_api.c:106

0x40033802: abort at D:/esp/esp-idf/components/newlib/abort.c:46

0x4002a52f: _esp_error_check_failed at D:/esp/esp-idf/components/esp_common/src/esp_err.c:42

0x4008bfc8: app_main at d:\esp\esp-qcloud-master\examples\led_light\build/../main/app_main.c:318 (discriminator 1)

0x4013877b: main_task at D:/esp/esp-idf/components/freertos/xtensa/port.c:537 (discriminator 2)

0x4002eb1d: vPortTaskWrapper at D:/esp/esp-idf/components/freertos/xtensa/port.c:170

ELF file SHA256: a7de6de0f43a41af

I (830) esp_core_dump_flash: Save core dump to flash...
I (836) esp_core_dump_elf: Found tasks: 7
I (841) esp_core_dump_flash: Erase flash 12288 bytes @ 0x3c0000
I (951) esp_core_dump_flash: Write end offset 0x26d8, check sum length 4

请问是不是qcloud-console使用的uart0和和我增加的uart1冲突造成这个错误呢？

isky111, 2021-02-18:
您好，我通过检查函数原型发现您似乎添加了 modbus 例程，我尝试将其添加到 esp-qcloud 测试，并未复现所述的情况，我所使用的 idf 为 release/v4.2，commit: 8cd16b。通过 log 判断，这个并非为 qcloud console 的问题，可以通过配置 `[] The device will be in debug mode`  关闭 console 功能加以验证。您可以提供所使用的 idf 版本，以及相关改动吗？我这边尝试复现一下。

jenking, 2021-02-18:
SP-ROM:esp32s2-rc4-20191025Build:Oct 25 2019rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)SPIWP:0xeemode:DIO, clock div:1load:0x3ffe6100,len:0x8load:0x3ffe6108,len:0x1904load:0x4004c000,len:0xc68load:0x40050000,len:0x2f94entry 0x4004c214I (46) boot: ESP-IDF v4.3-dev-1720-g494a124d9-dirty 2nd stage bootloaderI (46) boot: compile time 20:15:56I (46) boot: chip revision: 0I (50) qio_mode: Enabling default flash chip QIOI (55) boot.esp32s2: SPI Speed      : 80MHzI (60) boot.esp32s2: SPI Mode       : QIOI (64) boot.esp32s2: SPI Flash Size : 4MBI (69) boot: Enabling RNG early entropy source...I (75) boot: Partition Table:I (78) boot: ## Label            Usage          Type ST Offset   LengthI (85) boot:  0 nvs              WiFi data        01 02 0000d000 00008000I (93) boot:  1 fctry            WiFi data        01 02 00015000 00004000I (100) boot:  2 log_status       WiFi data        01 02 00019000 00004000I (108) boot:  3 otadata          OTA data         01 00 0001d000 00002000I (115) boot:  4 phy_init         RF data          01 01 0001f000 00001000I (123) boot:  5 ota_0            OTA app          00 10 00020000 001d0000I (130) boot:  6 ota_1            OTA app          00 11 001f0000 001d0000I (138) boot:  7 coredump         Unknown data     01 03 003c0000 00010000I (146) boot:  8 log_info         Unknown data     01 fe 003d0000 00010000I (153) boot:  9 reserved         Unknown data     01 ff 003e0000 00020000I (161) boot: End of partition tableI (165) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f000020 size=0x3029c (197276) mapI (210) esp_image: segment 1: paddr=0x000502c4 vaddr=0x3ffc8a20 size=0x04fcc ( 20428) loadI (214) esp_image: segment 2: paddr=0x00055298 vaddr=0x40024000 size=0x00404 (  1028) load0x40024000: _WindowOverflow4 at D:/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730I (217) esp_image: segment 3: paddr=0x000556a4 vaddr=0x40024404 size=0x0a974 ( 43380) load0x40024404: _coredump_iram_start at ??:?I (236) esp_image: segment 4: paddr=0x00060020 vaddr=0x40080020 size=0xb8ba0 (756640) map0x40080020: _stext at ??:?I (373) esp_image: segment 5: paddr=0x00118bc8 vaddr=0x4002ed78 size=0x09ca0 ( 40096) load0x4002ed78: prvCopyDataToQueue at D:/esp/esp-idf/components/freertos/queue.c:2123I (394) boot: Loaded app from partition at offset 0x20000I (394) boot: Disabling RNG early entropy source...I (406) cache: Instruction cache        : size 16KB, 4Ways, cache line size 16ByteI (406) cpu_start: Pro cpu up.I (462) cpu_start: Pro cpu start user codeI (462) cpu_start: cpu freq: 240000000I (462) cpu_start: Application information:I (465) cpu_start: Project name:     led_lightI (470) cpu_start: App version:      1I (474) cpu_start: Compile time:     Feb 16 2021 01:31:15I (480) cpu_start: ELF file SHA256:  f148e4dec07837ed...I (486) cpu_start: ESP-IDF:          v4.3-dev-1720-g494a124d9-dirtyI (493) heap_init: Initializing. RAM available for dynamic allocation:I (500) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAMI (507) heap_init: At 3FFD23A0 len 00029C60 (167 KiB): DRAMI (513) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAMI (519) spi_flash: detected chip: genericI (524) spi_flash: flash io: qioI (531) esp_core_dump_flash: Init core dump to flashI (533) esp_core_dump_flash: Found partition 'coredump' @ 3c0000 65536 bytesI (542) esp_core_dump_flash: Found core dump 15784 bytes in flash @ 0x3c0000I (549) cpu_start: Starting scheduler on PRO CPU.D (565) esp_qcloud_reboot: reboot unbroken count: 1D (566) esp_qcloud_reboot: version_fallback_task exitV (570) esp_qcloud_mem: <esp_qcloud_log : 162> Alloc ptr: 0x3ffd7dc4, size: 16, heap free: 159956W (579) esp_qcloud_log_flash: <ESP_ERR_NVS_NOT_FOUND> NVS open for log_status log_info log_info failedV (588) esp_qcloud_mem: <esp_qcloud_log_flash : 157> Alloc ptr: 0x3ffd8a2c, size: 24, heap free: 156800I (721) esp_qcloud_log_flash: LOG flash initialized successfullyI (721) esp_qcloud_log_flash: Log save partition subtype: label: log_info, addr:0x3d0000, offset: 0, size: 65536I (727) esp_qcloud_log: log initialized successfullyE (740) app_main: The device has been restarted abnormallyI (745) wifi:wifi driver task: 3ffdd574, prio:23, stack:6656, core=0I (745) system_api: Base MAC address is not setI (750) system_api: read default base MAC address from EFUSEI (760) wifi:wifi firmware version: d5c1f19I (761) wifi:wifi certification version: v7.0I (764) wifi:config NVS flash: enabledI (768) wifi:config nano formating: disabledI (772) wifi:Init data frame dynamic rx buffer num: 24I (777) wifi:Init management frame dynamic rx buffer num: 24I (782) wifi:Init management short buffer num: 32I (786) wifi:Init dynamic tx buffer num: 24I (790) wifi:Init static rx buffer size: 1600I (795) wifi:Init static rx buffer num: 8I (798) wifi:Init dynamic rx buffer num: 24I (803) wifi_init: rx ba win: 16I (806) wifi_init: tcpip mbox: 32I (810) wifi_init: udp mbox: 32I (814) wifi_init: tcp mbox: 32I (818) wifi_init: tcp tx win: 28000I (822) wifi_init: tcp rx win: 28000I (826) wifi_init: tcp mss: 1440I (830) wifi_init: WiFi IRAM OP enabledI (835) wifi_init: WiFi RX IRAM OP enabledI (965) phy: phy_version: 603, 72dfd77, Jul  7 2020, 19:57:05, 0, 2I (966) wifi:mode : sta (7c:df:a1:08:32:44)I (967) wifi:enable tsfI (1089) wifi:new:<3,0>, old:<1,0>, ap:<255,255>, sta:<3,0>, prof:1I (1090) wifi:state: init -> auth (b0)I (1094) wifi:state: auth -> assoc (0)I (1098) wifi:state: assoc -> run (10)W (1108) wifi:<ba-add>idx:0 (ifx:0, b4:0f:3b:64:c9:31), tid:0, ssn:0, win [... cut ...]  D:/esp/esp-idf/components/freertos/xtensa/port.c:537 (discriminator 2)0x4002eb1d: vPortTaskWrapper at D:/esp/esp-idf/components/freertos/xtensa/port.c:170ELF file SHA256: f148e4dec07837edI (4517) esp_core_dump_flash: Save core dump to flash...I (4523) esp_core_dump_elf: Found tasks: 11I (4528) esp_core_dump_flash: Erase flash 16384 bytes @ 0x3c0000I (4682) esp_core_dump_flash: Write end offset 0x3da0, check sum length 4I (4682) esp_core_dump_flash: Core dump has been saved to flash.Rebooting...

jenking, 2021-02-19:
[led_light.zip](https://github.com/espressif/esp-qcloud/files/6007072/led_light.zip)

jenking, 2021-02-19:
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x4
load:0x3ffe6104,len:0x19d8
load:0x4004c000,len:0x1a6c
load:0x40050000,len:0x2100
entry 0x4004c35c
I (46) boot: ESP-IDF v4.2-dirty 2nd stage bootloader
I (46) boot: compile time 11:00:33
I (46) boot: chip revision: 0
I (48) qio_mode: Enabling default flash chip QIO
I (53) boot.esp32s2: SPI Speed      : 80MHz
I (58) boot.esp32s2: SPI Mode       : QIO
I (63) boot.esp32s2: SPI Flash Size : 4MB
I (67) boot: Enabling RNG early entropy source...
I (73) boot: Partition Table:
I (76) boot: ## Label            Usage          Type ST Offset   Length
I (84) boot:  0 nvs              WiFi data        01 02 0000d000 00008000
I (91) boot:  1 fctry            WiFi data        01 02 00015000 00004000
I (99) boot:  2 log_status       WiFi data        01 02 00019000 00004000
I (106) boot:  3 otadata          OTA data         01 00 0001d000 00002000
I (114) boot:  4 phy_init         RF data          01 01 0001f000 00001000
I (121) boot:  5 ota_0            OTA app          00 10 00020000 001d0000
I (129) boot:  6 ota_1            OTA app          00 11 001f0000 001d0000
I (136) boot:  7 coredump         Unknown data     01 03 003c0000 00010000
I (144) boot:  8 log_info         Unknown data     01 fe 003d0000 00010000
I (151) boot:  9 reserved         Unknown data     01 ff 003e0000 00020000
I (159) boot: End of partition table
I (163) boot: No factory image, trying OTA 0
I (168) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f000020 size=0x31484 (201860) map
I (214) esp_image: segment 1: paddr=0x000514ac vaddr=0x3ffc95b0 size=0x04830 ( 18480) load
I (218) esp_image: segment 2: paddr=0x00055ce4 vaddr=0x40024000 size=0x00404 (  1028) load
0x40024000: _WindowOverflow4 at D:/esp/espidf42/esp-idf-v4.2/components/freertos/xtensa/xtensa_vectors.S:1730

I (221) esp_image: segment 3: paddr=0x000560f0 vaddr=0x40024404 size=0x09f28 ( 40744) load
I (239) esp_image: segment 4: paddr=0x00060020 vaddr=0x40080020 size=0xba19c (762268) map
0x40080020: _stext at ??:?

I (377) esp_image: segment 5: paddr=0x0011a1c4 vaddr=0x4002e32c size=0x0b278 ( 45688) load
0x4002e32c: xQueueGenericSendFromISR at D:/esp/espidf42/esp-idf-v4.2/components/freertos/queue.c:1162 (discriminator 1)

I (400) boot: Loaded app from partition at offset 0x20000
I (444) boot: Set actual ota_seq=1 in otadata[0]
I (445) boot: Disabling RNG early entropy source...
I (445) cache: Instruction cache        : size 16KB, 4Ways, cache line size 16Byte
I (452) cpu_start: Pro cpu up.
I (456) cpu_start: Application information:
I (460) cpu_start: Project name:     led_light
I (466) cpu_start: App version:      1
I (470) cpu_start: Compile time:     Feb 19 2021 11:00:16
I (476) cpu_start: ELF file SHA256:  5dfc231ecb8827a6...
I (482) cpu_start: ESP-IDF:          v4.2-dirty
I (487) cpu_start: Single core mode
I (492) heap_init: Initializing. RAM available for dynamic allocation:
I (499) heap_init: At 3FFD2FB8 len 00029048 (164 KiB): DRAM
I (505) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (511) cpu_start: Pro cpu start user code
I (573) spi_flash: detected chip: generic
I (573) spi_flash: flash io: qio
I (573) esp_core_dump_flash: Init core dump to flash
I (576) esp_core_dump_flash: Found partition 'coredump' @ 3c0000 65536 bytes
I (585) esp_core_dump_flash: Found core dump 15776 bytes in flash @ 0x3c0000
I (591) cpu_start: Starting scheduler on PRO CPU.
D (607) esp_qcloud_reboot: reboot unbroken count: 2
D (608) esp_qcloud_reboot: version_fallback_task exit
V (612) esp_qcloud_mem: <esp_qcloud_log : 162> Alloc ptr: 0x3ffd8fd8, size: 16, heap free: 158200
W (621) esp_qcloud_log_flash: <ESP_ERR_NVS_NOT_FOUND> NVS open for log_status log_info log_info failed
[... 79 lines cut ...]
0x4002a367: _esp_error_check_failed at D:/esp/espidf42/esp-idf-v4.2/components/esp_common/src/esp_err.c:46

Backtrace:0x4002b4a7:0x3ffd6ab0 0x4002bd25:0x3ffd6ad0 0x400322ee:0x3ffd6af0 0x4002a367:0x3ffd6b60 0x400887b0:0x3ffd6b80 0x40086917:0x3ffd6c60 0x4002bd2d:0x3ffd6c90
0x4002b4a7: panic_abort at D:/esp/espidf42/esp-idf-v4.2/components/esp_system/panic.c:330

0x4002bd25: esp_system_abort at D:/esp/espidf42/esp-idf-v4.2/components/esp_system/system_api.c:106

0x400322ee: abort at D:/esp/espidf42/esp-idf-v4.2/components/newlib/abort.c:46

0x4002a367: _esp_error_check_failed at D:/esp/espidf42/esp-idf-v4.2/components/esp_common/src/esp_err.c:46

0x400887b0: app_main at d:\esp\esp-qcloud-master\examples\led_light\build/../main/app_main.c:343 (discriminator 1)

0x40086917: main_task at D:/esp/espidf42/esp-idf-v4.2/components/esp32s2/cpu_start.c:453 (discriminator 2)

0x4002bd2d: vPortTaskWrapper at D:/esp/espidf42/esp-idf-v4.2/components/freertos/xtensa/port.c:143

ELF file SHA256: 5dfc231ecb8827a6

I (3629) esp_core_dump_flash: Save core dump to flash...
I (3635) esp_core_dump_elf: Found tasks: 11
I (3640) esp_core_dump_flash: Erase flash 16384 bytes @ 0x3c0000
I (3796) esp_core_dump_flash: Write end offset 0x3da8, check sum length 4
I (3796) esp_core_dump_flash: Core dump has been saved to flash.
Rebooting...

isky111, 2021-02-19:
我复现出了你的问题，`QCloud` 的 `led` 组件使用了**定时器组 0 的 T0 定时器**，这与 `modbus` 所使用的定时器冲突，更改下 `modbus` 定时器即可。
``` 
(Top) → Component config → Modbus configuration
(0) Modbus Timer group number
(1) Modbus Timer index in the grou
```

jenking, 2021-02-19:
你好，问题已经解决，的确是计时器出现的问题，请问您是如何发现这个问题的呢？

closed 2021-02-21 by jenking