espressif/esp-qcloud #11: 你好，能不能提供一个使用action的例子

jenking, 2021-02-28:
你好，例程里只有属性是例子，能不能出一个如何使用action的例子呢？谢谢

shixinke-orion, 2021-03-01:
您好，步骤如下。
1. 通过  `esp_qcloud_device_add_action_cb` 注册 `action` 的回调函数。
    回调函数的原型为 `typedef esp_err_t (*esp_qcloud_action_cb_t)(esp_qcloud_method_t *action_handle, char *params)`
2. 云下发参数通过回调函数的 `params` 获取，如需上报参数直接使用 `esp_qcloud_iothub_param_add_x` 添加即可。

使用示例：
```
esp_err_t open_action_cb(esp_qcloud_method_t *action_handle, char *params)
{
    //云下发的参数
    ESP_LOGI(TAG, "params： %s", params);
    
    //上报参数
    esp_qcloud_iothub_param_add_bool(action_handle， "status", 1);

   //返回执行结果，该结果将上报云。
    return ESP_OK;
}

esp_qcloud_device_add_action_cb("open", open_action_cb);
```

jenking, 2021-03-03:
light_driver.h:365:23: error: unknown type name 'esp_qcloud_method_t'
 esp_err_t action_feed(esp_qcloud_method_t *action_handle, int *params);
                       ^~~~~~~~~~~~~~~~~~~
../main/light_driver.c:815:23: error: unknown type name 'esp_qcloud_method_t'; did you mean 'esp_qcloud_time_config_t'?
 esp_err_t action_feed(esp_qcloud_method_t *action_handle, int *params)
                       ^~~~~~~~~~~~~~~~~~~
                       esp_qcloud_time_config_t
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

请问这个是什么问题？

shixinke-orion, 2021-03-03:
`esp_qcloud_method_t ` 被定义在  `esp_qcloud_iothub.h`，检查是否包含此头文件。若未发现此定义，可通过 `git pull` 拉取最新的代码。

jenking, 2021-03-03:
已经手动在light_driver中include这个头文件，已经是最新的代码。还是存在这个问题

shixinke-orion, 2021-03-03:
`action` 功能 commit: 538b1103742d0bab0c75477963e0849dc326cd94，通过 git show 检查下本地仓库是否存在 该提交。

jenking, 2021-03-03:
你好，我之前看过你们提交代码的记录，我也查看过本地的代码，我的iothub.c里面的代码有这句
typedef esp_err_t (*esp_qcloud_action_cb_t)(esp_qcloud_method_t *action_handle, char *params)

以及esp_qcloud_device_add_action_cb这些新增加的函数都有

shixinke-orion, 2021-03-03:
可以尝试删除 `build 文件` 重新构建工程。

jenking, 2021-03-03:
更新代码之后，我第一时间就是idf.py fullclean

shixinke-orion, 2021-03-04:
可以提供工程或相关文件吗？我可以帮您 `review` 下。

jenking, 2021-03-04:
你好，我刚刚又重新下载源码，什么都没改动，直接编译例程序，也是出现这个问题。

In file included from D:/esp/esp-qcloud-master/src/iothub/esp_qcloud_iothub.c:2
:
../../../include/esp_qcloud_iothub.h:149:28: error: field 'action_cb' declared as a function
     esp_qcloud_action_cb_t action_cb;
                            ^~~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

jenking, 2021-03-04:
你好，重新修改代码后，已解决

closed 2021-03-04 by jenking