espressif/esp-qcloud #12: 使用esp_qcloud_iothub_report_all_property上报属性不全

aixier, 2021-03-12:
注册属性：
    ESP_ERROR_CHECK(esp_qcloud_device_add_property("bg_num", QCLOUD_VAL_TYPE_INTEGER));
    ESP_ERROR_CHECK(esp_qcloud_device_add_property("customer_num", QCLOUD_VAL_TYPE_INTEGER));
	ESP_ERROR_CHECK(esp_qcloud_device_add_property("dt_progress", QCLOUD_VAL_TYPE_INTEGER));
    **ESP_ERROR_CHECK(esp_qcloud_device_add_property("check", QCLOUD_VAL_TYPE_ENUM));**
    ESP_ERROR_CHECK(esp_qcloud_device_add_property("power", QCLOUD_VAL_TYPE_BOOLEAN));
    ESP_ERROR_CHECK(esp_qcloud_device_add_property("temperature", QCLOUD_VAL_TYPE_INTEGER));
    ESP_ERROR_CHECK(esp_qcloud_device_add_property("add", QCLOUD_VAL_TYPE_BOOLEAN));
    ESP_ERROR_CHECK(esp_qcloud_device_add_property("sub", QCLOUD_VAL_TYPE_BOOLEAN));
	ESP_ERROR_CHECK(esp_qcloud_device_add_property("auto", QCLOUD_VAL_TYPE_BOOLEAN));
    /**< The processing function of the communication between the device and the serve
上报属性：
thing/up/property/VR9I8G3GN0/customer_check, method: report, data: {"method":"report","clientToken":"customer_check-34090","params":{"bg_num":0,"auto":1,"sub":1,"add":1,"temperature":97,"power":1,"dt_progress":0,"customer_num":0}}[0m
[0;32mI (9399121) esp_qcloud_iothub: property_callback, topic:

shixinke-orion, 2021-03-12:
`esp_qcloud_iothub_report_all_property` 使用 `esp_qcloud_device_add_property_cb` 所注册的回调函数获取参数。该回调由 `esp_qcloud_handle_get_param` 所触发，他将检查所有由 `esp_qcloud_device_add_property` 所添加的 `property`。
- 请检查你所注册的回调是否处理了所有 `get` 请求。
- 在`esp_qcloud_handle_get_param` 增加打印可确定是否遍历了所有 `property`。
- 检查所有注册的`property`的类型是否与云端一致。

aixier, 2021-03-26:
NP

closed 2021-03-26 by aixier