espressif/esp-qcloud #14: Does not support this authentication method,

aixier, 2021-06-08:
start_Countdown: failed!
esp_timer_start_periodic: ok!
[0;32mI (5640) esp_qcloud_timesync: Initializing SNTP. Using the SNTP server: pool.ntp.org[0m
[0;31mE (5649) esp_qcloud_iothub: Does not support this authentication method, auth_mode: 0[0m
[0;32mI (5657) esp_qcloud_mqtt: Initialising MQTT[0m
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400014fd  PS      : 0x00060130  A0      : 0x801c3ad8  A1      : 0x3ffba3c0  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffdad4e  
A10     : 0x3ffdad3d  A11     : 0x3f404ed0  A12     : 0x00000011  A13     : 0x00000000  
A14     : 0x3ffdfff8  A15     : 0x3ffdad30  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

shixinke-orion, 2021-06-09:
Did you use `esp_qcloud_create_device` to create the device? Currently, only KEY verification is supported for mass production customers.

aixier, 2021-06-14:
TKS

closed 2021-06-14 by aixier