espressif/esp-qcloud #15: How to solve the following problems?

modi12jin, 2021-06-12:
[image]

```
/home/jin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/esp-qcloud/libesp-qcloud.a(esp_qcloud_prov_ble.o):(.literal.esp_qcloud_prov_ble_handle_recv_data+0x34): undefined reference to `esp_blufi_send_custom_data'
/home/jin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/esp-qcloud/libesp-qcloud.a(esp_qcloud_prov_ble.o):(.literal.ble_event_callback+0x64): undefined reference to `esp_blufi_close'
/home/jin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/esp-qcloud/libesp-qcloud.a(esp_qcloud_prov_ble.o):(.literal.esp_qcloud_prov_bleconfig_start+0x4c): undefined reference to `esp_blufi_register_callbacks'
/home/jin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/esp-qcloud/libesp-qcloud.a(esp_qcloud_prov_ble.o):(.literal.esp_qcloud_prov_bleconfig_start+0x50): undefined reference to `esp_blufi_profile_init'
/home/jin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/esp-qcloud/libesp-qcloud.a(esp_qcloud_prov_ble.o):(.literal.esp_qcloud_prov_ble_stop+0x1c): undefined reference to `esp_blufi_profile_deinit'
/home/jin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/esp-qcloud/libesp-qcloud.a(esp_qcloud_prov_ble.o): in function `esp_qcloud_prov_ble_handle_recv_data':
/home/jin/Desktop/ESP32/components/esp-qcloud/src/provisioning/esp_qcloud_prov_ble.c:126: undefined reference to `esp_blufi_send_custom_data'
/home/jin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/esp-qcloud/libesp-qcloud.a(esp_qcloud_prov_ble.o): in function `ble_event_callback':
/home/jin/Desktop/ESP32/components/esp-qcloud/src/provisioning/esp_qcloud_prov_ble.c:203: undefined reference to `esp_blufi_close'
/home/jin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/esp-qcloud/libesp-qcloud.a(esp_qcloud_prov_ble.o): in function `esp_qcloud_prov_ble_report_bind_status':
/home/jin/Desktop/ESP32/components/esp-qcloud/src/provisioning/esp_qcloud_prov_ble.c:157: undefined reference to `esp_blufi_send_custom_data'
/home/jin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/esp-qcloud/libesp-qcloud.a(esp_qcloud_prov_ble.o): in function `esp_qcloud_prov_bleconfig_start':
/home/jin/Desktop/ESP32/components/esp-qcloud/src/provisioning/esp_qcloud_prov_ble.c:270: undefined reference to `esp_blufi_register_callbacks'
/home/jin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/jin/Desktop/ESP32/components/esp-qcloud/src/provisioning/esp_qcloud_prov_ble.c:273: undefined reference to `esp_blufi_profile_init'
/home/jin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/esp-qcloud/libesp-qcloud.a(esp_qcloud_prov_ble.o): in function `esp_qcloud_prov_ble_stop':
/home/jin/Desktop/ESP32/components/esp-qcloud/src/provisioning/esp_qcloud_prov_ble.c:279: undefined reference to `esp_blufi_profile_deinit'
/home/jin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/esp-qcloud/libesp-qcloud.a(esp_qcloud_prov_tencent.o):(.literal.event_handler+0x18): undefined reference to `esp_blufi_send_wifi_conn_report'
/home/jin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/esp-qcloud/libesp-qcloud.a(esp_qcloud_prov_tencent.o): in function `event_handler':
/home/jin/Desktop/ESP32/components/esp-qcloud/src/provisioning/esp_qcloud_prov_tencent.c:92: undefined reference to `esp_blufi_send_wifi_conn_report'
/home/jin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/jin/Desktop/ESP32/components/esp-qcloud/src/provisioning/esp_qcloud_prov_tencent.c:104: undefined reference to `esp_blufi_send_wifi_conn_report'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1
```

shixinke-orion, 2021-06-15:
需要使用最新的 `release/v4.2`及以上版本，同时在 `menuconfig` 中开启下述选项
```
(Top) → Component config → Bluetooth → Bluedroid Options
[*]         Include blufi function
[*] Enable BLE 4.2 features
```

modi12jin, 2021-06-15:
[image]

thank you very much

closed 2021-06-15 by modi12jin