espressif/esp-qcloud #19: esp32c3 Build Error esp_blufi_close  [-Werror=implicit-function-declaration]

skylli, 2021-11-04:
/home/sky/esp/prj/esp-qcloud/src/provisioning/esp_qcloud_prov_ble.c: In function 'ble_event_callback':
/home/sky/esp/prj/esp-qcloud/src/provisioning/esp_qcloud_prov_ble.c:208:9: error: implicit declaration of function 'esp_blufi_close'; did you mean 'esp_wifi_stop'? [-Werror=implicit-function-declaration]
         esp_blufi_close(server_if, conn_id);
         ^~~~~~~~~~~~~~~
         esp_wifi_stop
cc1: some warnings being treated as errors
[1277/1305] Building C object esp-idf/esp-qcloud/CMakeFile...esp-qcloud.dir/src/provisioning/qrcode/src/qrcodegen.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

shixinke-orion, 2021-11-08:
在使用 `release/v4.3` 及以上版本时需要在`esp_qcloud_prov_ble.c`中加入`#include "esp_blufi.h"`

xupenghu, 2021-11-10:
编译报错
```
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c:63:46: error: 'timg_txconfig_reg_t' {aka 'volatile union <anonymous>'} has no member named 'enable'
     TG[group_num]->hw_timer[timer_num].config.enable = 0;
                                              ^
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c: In function 'fade_timercb':
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c:276:44: error: incompatible types when assigning to type 'timg_txupdate_reg_t' {aka 'volatile union <anonymous>'} from type 'int'
         TIMERG0.hw_timer[timer_idx].update = 1;
                                            ^
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c:285:35: error: 'timg_int_clr_timers_reg_t' {aka 'volatile union <anonymous>'} has no member named 't0'
             TIMERG0.int_clr_timers.t0 = 1;
                                   ^
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c:293:35: error: 'timg_int_clr_timers_reg_t' {aka 'volatile union <anonymous>'} has no member named 't1'
             TIMERG0.int_clr_timers.t1 = 1;
                                   ^
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c:302:44: error: 'timg_txconfig_reg_t' {aka 'volatile union <anonymous>'} has no member named 'alarm_en'; did you 
mean 'tx_alarm_en'?
         TIMERG0.hw_timer[timer_idx].config.alarm_en = TIMER_ALARM_EN;
                                            ^~~~~~~~
                                            tx_alarm_en
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c:308:44: error: incompatible types when assigning to type 'timg_txupdate_reg_t' {aka 'volatile union <anonymous>'} from type 'int'
         TIMERG1.hw_timer[timer_idx].update = 1;
                                            ^
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c:316:35: error: 'timg_int_clr_timers_reg_t' {aka 'volatile union <anonymous>'} has no member named 't0'
             TIMERG1.int_clr_timers.t0 = 1;
                                   ^
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c:322:35: error: 'timg_int_clr_timers_reg_t' {aka 'volatile union <anonymous>'} has no member named 't1'
             TIMERG1.int_clr_timers.t1 = 1;
                                   ^
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c:328:44: error: 'timg_txconfig_reg_t' {aka 'volatile union <anonymous>'} has no member named 'alarm_en'; did you 
mean 'tx_alarm_en'?
         TIMERG1.hw_timer[timer_idx].config.alarm_en = TIMER_ALARM_EN;
                                            ^~~~~~~~
                                            tx_alarm_en
[1302/1311] Building C object esp-idf/esp-qcloud/CMakeFiles/__idf_esp-qcloud.dir/src/utils/esp_qcloud_utils.c.obj
```