espressif/esp-qcloud #21: 编译报错

xupenghu, 2021-11-10:
```
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c:63:46: error: 'timg_txconfig_reg_t' {aka 'volatile union <anonymous>'} has no member named 'enable'
     TG[group_num]->hw_timer[timer_num].config.enable = 0;
                                              ^
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c: In function 'fade_timercb':
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c:276:44: error: incompatible types when assigning to type 'timg_txupdate_reg_t' {aka 'volatile union <anonymous>'} from type 'int'
         TIMERG0.hw_timer[timer_idx].update = 1;
                                            ^
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c:285:35: error: 'timg_int_clr_timers_reg_t' {aka 'volatile union <anonymous>'} has no member named 't0'
             TIMERG0.int_clr_timers.t0 = 1;
                                   ^
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c:293:35: error: 'timg_int_clr_timers_reg_t' {aka 'volatile union <anonymous>'} has no member named 't1'
             TIMERG0.int_clr_timers.t1 = 1;
                                   ^
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c:302:44: error: 'timg_txconfig_reg_t' {aka 'volatile union <anonymous>'} has no member named 'alarm_en'; did you 
mean 'tx_alarm_en'?
         TIMERG0.hw_timer[timer_idx].config.alarm_en = TIMER_ALARM_EN;
                                            ^~~~~~~~
                                            tx_alarm_en
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c:308:44: error: incompatible types when assigning to type 'timg_txupdate_reg_t' {aka 'volatile union <anonymous>'} from type 'int'
         TIMERG1.hw_timer[timer_idx].update = 1;
                                            ^
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c:316:35: error: 'timg_int_clr_timers_reg_t' {aka 'volatile union <anonymous>'} has no member named 't0'
             TIMERG1.int_clr_timers.t0 = 1;
                                   ^
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c:322:35: error: 'timg_int_clr_timers_reg_t' {aka 'volatile union <anonymous>'} has no member named 't1'
             TIMERG1.int_clr_timers.t1 = 1;
                                   ^
/mnt/d/UGit/esp/esp-qcloud/examples/common_components/led/src/led_pwm.c:328:44: error: 'timg_txconfig_reg_t' {aka 'volatile union <anonymous>'} has no member named 'alarm_en'; did you 
mean 'tx_alarm_en'?
         TIMERG1.hw_timer[timer_idx].config.alarm_en = TIMER_ALARM_EN;
                                            ^~~~~~~~
                                            tx_alarm_en
[1302/1311] Building C object esp-idf/esp-qcloud/CMakeFiles/__idf_esp-qcloud.dir/src/utils/esp_qcloud_utils.c.obj
```

shixinke-orion, 2021-11-15:
需要使用 `release/v4.2` 及以上编译。如果还是无法编译通过，提供下你所使用的的版本及 `commit`。

xupenghu, 2021-11-16:
> 需要使用 `release/v4.2` 及以上编译。如果还是无法编译通过，提供下你所使用的的版本及 `commit`。

使用了master分支代码 commit id ： bcbef9a8db54d2deef83402f6e4403ccf298803a

closed 2022-03-01 by xupenghu

W00ng, 2022-07-31:
我这边在4.4也遇到类似报错，使用esp32-c3，请问如何解决了