espressif/esp-qcloud #28: 使用idf ralease/4.3编译esp32c3成功，但是一直重启，我的log如下

Qiu-zhangbao, 2022-07-18:
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1838
load:0x403ce000,len:0x8d4
load:0x403d0000,len:0x2d80
entry 0x403ce000
I (35) boot: ESP-IDF v4.3.3-98-g9bbe2fc041 2nd stage bootloader
I (35) boot: compile time 15:02:44
I (35) boot: chip revision: 3
I (38) boot.esp32c3: SPI Speed      : 80MHz
I (43) boot.esp32c3: SPI Mode       : DIO
I (48) boot.esp32c3: SPI Flash Size : 4MB
W (52) boot.esp32c3: PRO CPU has been reset by WDT.
E (58) boot.esp32c3: WDT reset info dump is not supported yet
I (64) boot: Enabling RNG early entropy source...
I (70) boot: Partition Table:
I (73) boot: ## Label            Usage          Type ST Offset   Length
I (81) boot:  0 nvs              WiFi data        01 02 0000d000 00008000
I (88) boot:  1 fctry            WiFi data        01 02 00015000 00004000
I (95) boot:  2 log_status       WiFi data        01 02 00019000 00004000
I (103) boot:  3 otadata          OTA data         01 00 0001d000 00002000
I (110) boot:  4 phy_init         RF data          01 01 0001f000 00001000
I (118) boot:  5 ota_0            OTA app          00 10 00020000 001d0000
I (126) boot:  6 ota_1            OTA app          00 11 001f0000 001d0000
I (133) boot:  7 coredump         Unknown data     01 03 003c0000 00010000
I (141) boot:  8 log_info         Unknown data     01 fe 003d0000 00010000
I (148) boot:  9 reserved         Unknown data     01 ff 003e0000 00020000
I (156) boot: End of partition table
I (160) esp_image: segment 0: paddr=00020020 vaddr=3c130020 size=44888h (280712)
 map
I (213) esp_image: segment 1: paddr=000648b0 vaddr=3fc90800 size=03c94h ( 15508)
 load
I (216) esp_image: segment 2: paddr=0006854c vaddr=40380000 size=07acch ( 31436)
 load
I (225) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=126114h (120450
0) map
I (420) esp_image: segment 4: paddr=0019613c vaddr=40387acc size=08cd8h ( 36056)
 load
I (427) esp_image: segment 5: paddr=0019ee1c vaddr=50000010 size=00010h (    16)
 load
I (432) boot: Loaded app from partition at offset 0x20000
I (432) boot: Disabling RNG early entropy source...
I (448) cpu_start: Pro cpu up.
I (461) cpu_start: Pro cpu start user code
I (461) cpu_start: cpu freq: 160000000
I (461) cpu_start: Application information:
I (464) cpu_start: Project name:     led_light
I (469) cpu_start: App version:      1
I (473) cpu_start: Compile time:     Jul 18 2022 15:02:40
I (479) cpu_start: ELF file SHA256:  e5b5e8dda4963fc0...
I (485) cpu_start: ESP-IDF:          v4.3.3-98-g9bbe2fc041
I (491) heap_init: Initializing. RAM available for dynamic allocation:
I (499) heap_init: At 3FC9BD00 len 00024300 (144 KiB): DRAM
I (505) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (512) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (518) spi_flash: detected chip: mxic
I (522) spi_flash: flash io: dio
I (527) sleep: Configure to isolate all GPIO pins in sleep state
I (533) sleep: Enable automatic switching of GPIO sleep configuration
I (540) cpu_start: Starting scheduler.
D (561) esp_qcloud_reboot: reboot unbroken count: 141
E (564) esp_ota_ops: Rollback is not possible, do not have any suitable apps in
slots
D (566) esp_qcloud_reboot: version_fallback_task exit
W (574) esp_qcloud_log_flash: <ESP_ERR_NVS_NOT_FOUND> NVS open for log_status lo
g_info log_info failed
I (892) esp_qcloud_log_flash: LOG flash initialized successfully
I (892) esp_qcloud_log_flash: Log save partition subtype: label: log_info, addr:
0x3d0000, offset: 0, size: 65536
I (898) esp_qcloud_log: log initialized successfully
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
0x40380082: _panic_handler at vectors.S.obj:?

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1838
load:0x403ce000,len:0x8d4
load:0x403d0000,len:0x2d80
entry 0x403ce000

[image]

shixinke-orion, 2022-07-19:
参考该 issue 解决方案即可，https://github.com/espressif/esp-qcloud/issues/17

Qiu-zhangbao, 2022-07-21:
> 建议切换到 v4.3.2 测试，其他分支并未适配。我在 v4.3.2 上测试 BLE 与 softap 是没问题的。 对于 BLE 配网你需要扫描 shell 终端出来的二维码或者产品开发中 第 5 步 `批量投产` 页面的二维码。

是否有适配idf4.4.1或者idf5.0的打算，这两个才是稳定版本，这个idf4.3.2是否可以用于量产？