espressif/esp-qcloud #30: 请问有支持esp32s3的计划吗？

neverMorexxxx, 2022-08-16:
代码移植到esp32s3 使用idf-4.4.2，无法正常运行，请问后续有支持esp32s3的计划吗？

shixinke-orion, 2022-08-16:
有计划。可以提供下出错 log 嘛？

Szeroy, 2022-08-17:
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 473.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory d:\workproject\workproject\outstandingproject\hand_held_air_detector
Executing "D:\esp-idf\esp-tools-4.4v\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe D:\esp-idf\esp-idf-4.4v\tools/idf_monitor.py -p com4 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 d:\workproject\workproject\outstandingproject\hand_held_air_detector\build\i80_controller.elf -m 'D:\esp-idf\esp-tools-4.4v\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe' 'D:\esp-idf\esp-idf-4.4v\tools\idf.py'"...
--- idf_monitor on com4 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1650
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2f88
entry 0x403c9954
I (24) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (25) boot: compile time 09:01:33
I (25) boot: chip revision: 0
I (26) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 2MB
I (40) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (83) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=29980h (170368) map
I (122) esp_image: segment 1: paddr=000399a8 vaddr=3fc95270 size=04204h ( 16900) load
I (126) esp_image: segment 2: paddr=0003dbb4 vaddr=40374000 size=02464h (  9316) load
I (130) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=ad67ch (710268) map
I (264) esp_image: segment 4: paddr=000ed6a4 vaddr=40376464 size=0ee04h ( 60932) load
I (278) esp_image: segment 5: paddr=000fc4b0 vaddr=50000000 size=00010h (    16) load
I (285) boot: Loaded app from partition at offset 0x10000
I (285) boot: Disabling RNG early entropy source...
I (298) cpu_start: Pro cpu up.
I (298) cpu_start: Starting app cpu, entry point is 0x403751ec
0x403751ec: call_start_cpu1 at D:/esp-idf/esp-idf-4.4v/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (312) cpu_start: Pro cpu start user code
I (312) cpu_start: cpu freq: 160000000
I (312) cpu_start: Application information:
I (315) cpu_start: Project name:     i80_controller
I (320) cpu_start: App version:      229152f-dirty
I (326) cpu_start: Compile time:     Aug 17 2022 09:01:14
I (332) cpu_start: ELF file SHA256:  ee8ba4cc03021ac2...
I (338) cpu_start: ESP-IDF:          v4.4.2
I (343) heap_init: Initializing. RAM available for dynamic allocation:
I (350) heap_init: At 3FC9D210 len 00042DF0 (267 KiB): D/IRAM
I (356) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (363) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (369) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (376) spi_flash: detected chip: gd
I (380) spi_flash: flash io: dio
W (384) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (397) sleep: Configure to isolate all GPIO pins in sleep state
I (404) sleep: Enable automatic switching of GPIO sleep configuration
I (411) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

Szeroy, 2022-08-17:
下载完成后，运行时卡在Starting scheduler on APP CPU.这个位置，我们的源码如下：

/* Event handler for catching QCloud events */
static void event_handler(void *arg, esp_event_base_t event_base,
                          int32_t event_id, void *event_data)
{
    switch (event_id)
    {
    case QCLOUD_EVENT_IOTHUB_INIT_DONE:
        esp_qcloud_iothub_report_device_info();
        ESP_LOGI(TAG, "QCloud Initialised");
        break;

    case QCLOUD_EVENT_IOTHUB_BOND_DEVICE:
        ESP_LOGI(TAG, "Device binding successful");
        break;

    case QCLOUD_EVENT_IOTHUB_UNBOND_DEVICE:
        ESP_LOGW(TAG, "Device unbound with iothub");
        esp_qcloud_storage_erase(CONFIG_QCLOUD_NVS_NAMESPACE);
        esp_restart();
        break;

    case QCLOUD_EVENT_IOTHUB_BIND_EXCEPTION:
        ESP_LOGW(TAG, "Device bind fail");
        esp_qcloud_storage_erase(CONFIG_QCLOUD_NVS_NAMESPACE);
        esp_restart();
        break;

    case QCLOUD_EVENT_IOTHUB_RECEIVE_STATUS:
        ESP_LOGI(TAG, "receive status message: %s", (char *)event_data);
        break;

    default:
        ESP_LOGW(TAG, "Unhandled QCloud Event: %d", event_id);
    }
}

static esp_err_t get_wifi_config(wifi_config_t *wifi_cfg, uint32_t wait_ms)
{
    esp_err_t ret = ESP_FAIL;

    ESP_QCLOUD_PARAM_CHECK(wifi_cfg);

    if (esp_qcloud_storage_get("wifi_config", wifi_cfg, sizeof(wifi_config_t)) == ESP_OK)
    {
        return ESP_OK;
    }
    // run_data.wifi_state = CONFIGURATION;

    ESP_ERROR_CHECK(esp_wifi_start());

    // while(ret != ESP_OK)
    {
        esp_qcloud_prov_softapconfig_start(SOFTAPCONFIG_TYPE_TENCENT, SOFT_AP_NAME, "");
        esp_qcloud_prov_print_wechat_qr(SOFT_AP_NAME, "softap");

        ret = esp_qcloud_prov_wait(wifi_cfg, wait_ms);

        esp_qcloud_prov_softapconfig_stop();
        if (ret == ESP_FAIL)
        {
            while (1)
            {
                vTaskDelay(1000);
            }
        }
    }

    // Store the configure of the device
    esp_qcloud_storage_set("wifi_config", wifi_cfg, sizeof(wifi_config_t));

    return ESP_OK;
}

void Qcloud_Task(void)
{
    esp_qcloud_log_config_t log_config = {
        .log_level_uart = ESP_LOG_INFO,
    };
    ESP_ERROR_CHECK(esp_qcloud_log_init(&log_config));
    esp_log_level_set("*", ESP_LOG_INFO);

    /**< Create and configure device authentication information */
    ESP_ERROR_CHECK(esp_qcloud_create_device());
    /**< Configure the version of the device, and use this information to determine whether to OTA */
    ESP_LOGI(TAG, "------>");
    ESP_ERROR_CHECK(esp_qcloud_device_add_fw_version("4.4v"));
    /**< Register the properties of the device */
    ESP_ERROR_CHECK(esp_qcloud_device_add_property("start_time", QCLOUD_VAL_TYPE_INTEGER));
    ESP_ERROR_CHECK(esp_qcloud_device_add_property("work_time", QCLOUD_VAL_TYPE_INTEGER));
    ESP_ERROR_CHECK(esp_qcloud_device_add_property("work_mode", QCLOUD_VAL_TYPE_INTEGER));
    ESP_ERROR_CHECK(esp_qcloud_device_add_property("grade", QCLOUD_VAL_TYPE_INTEGER));
    ESP_ERROR_CHECK(esp_qcloud_device_add_property("power", QCLOUD_VAL_TYPE_BOOLEAN));
    /**< The processing function of the communication between the device and the server */
    // ESP_ERROR_CHECK(esp_qcloud_device_add_property_cb(spot_get_param, spot_set_param));

    // /**
    //  * @brief Initialize Wi-Fi.
    //  */
     ESP_ERROR_CHECK(esp_qcloud_wifi_init());
     ESP_ERROR_CHECK(esp_event_handler_register(QCLOUD_EVENT, ESP_EVENT_ANY_ID, &event_handler, NULL));

    /**
     * @brief Get the router configuration
     */
     wifi_config_t wifi_cfg = {0};
     ESP_ERROR_CHECK(get_wifi_config(&wifi_cfg, portMAX_DELAY));

    // /**
    //  * @brief Connect to router
    //  */
     ESP_ERROR_CHECK(esp_qcloud_wifi_start(&wifi_cfg));
     ESP_ERROR_CHECK(esp_qcloud_timesync_start());

   // IOT_DynReg_Device();
    /*������ʱ���ܵ���mqtt��һ��connectʧ��*/
    vTaskDelay(500 / portTICK_PERIOD_MS);
    /**
     * @brief Connect to Tencent Cloud Iothub
     */
    ESP_ERROR_CHECK(esp_qcloud_iothub_init());
    ESP_ERROR_CHECK(esp_qcloud_iothub_start());
    ESP_ERROR_CHECK(esp_qcloud_iothub_ota_enable());

    vTaskDelete(NULL);
}

    void app_main(void)
{
    printf("================== APPINFO ===================\n");
    printf("    ESP_IDF VERSION: %s\n", esp_get_idf_version());
    printf("    APP VERSION: %s\n", "4.4v");
    printf("    Compile time: %s %s\n", __DATE__, __TIME__);
    printf("================================================\n");

    xTaskCreate(Qcloud_Task, "Qcloud_Task", 1024 * 10, NULL, 7, NULL);

    while (1)
    {
        printf("\n\n-------------------------------- Get Systrm Info------------------------------------------\n");
        //��ȡIDF�汾
        printf("    SDK version:%s\n", esp_get_idf_version());
        //��ȡоƬ�����ڴ�
        printf("    esp_get_free_heap_size : %d  \n", esp_get_free_heap_size());
        //��ȡ��δʹ�ù�����С�ڴ�
        printf("    esp_get_minimum_free_heap_size : %d  \n", esp_get_minimum_free_heap_size());
        // play_wt588f02a_audio(0);
        vTaskDelay(5000 / portTICK_PERIOD_MS);
    }
}

shixinke-orion, 2022-08-17:
我这边测试修改了下运行没啥问题，可能跟你开的双核模式有关。 下面是我移植到 v4.4.2 启动的 log
```
I (29) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (29) boot: compile time 16:15:49
I (29) boot: chip revision: 0
I (31) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32s3: Boot SPI Speed : 80MHz
I (41) boot.esp32s3: SPI Mode       : QIO
I (46) boot.esp32s3: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 0000d000 00008000
I (74) boot:  1 fctry            WiFi data        01 02 00015000 00004000
I (81) boot:  2 log_status       WiFi data        01 02 00019000 00004000
I (89) boot:  3 otadata          OTA data         01 00 0001d000 00002000
I (96) boot:  4 phy_init         RF data          01 01 0001f000 00001000
I (104) boot:  5 ota_0            OTA app          00 10 00020000 001d0000
I (111) boot:  6 ota_1            OTA app          00 11 001f0000 001d0000
I (119) boot:  7 coredump         Unknown data     01 03 003c0000 00010000
I (127) boot:  8 log_info         Unknown data     01 fe 003d0000 00010000
I (134) boot:  9 reserved         Unknown data     01 ff 003e0000 00020000
I (142) boot: End of partition table
I (146) esp_image: segment 0: paddr=00020020 vaddr=3c0c0020 size=2a088h (172168) map
I (181) esp_image: segment 1: paddr=0004a0b0 vaddr=3fc950f0 size=05664h ( 22116) load
I (185) esp_image: segment 2: paddr=0004f71c vaddr=40374000 size=008fch (  2300) load
I (187) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=b0038h (720952) map
I (304) esp_image: segment 4: paddr=00100060 vaddr=403748fc size=107f0h ( 67568) load
I (317) esp_image: segment 5: paddr=00110858 vaddr=50000000 size=00010h (    16) load
I (325) boot: Loaded app from partition at offset 0x20000
I (325) boot: Disabling RNG early entropy source...
I (337) cpu_start: Pro cpu up.
I (337) cpu_start: Single core mode
I (346) cpu_start: Pro cpu start user code
I (346) cpu_start: cpu freq: 160000000
I (346) cpu_start: Application information:
I (349) cpu_start: Project name:     led_light
I (354) cpu_start: App version:      43e218d-dirty
I (359) cpu_start: Compile time:     Aug 17 2022 16:15:44
I (366) cpu_start: ELF file SHA256:  b092d79df35e585f...
I (372) cpu_start: ESP-IDF:          v4.4.2
I (376) heap_init: Initializing. RAM available for dynamic allocation:
I (384) heap_init: At 3FC9E948 len 000416B8 (261 KiB): D/IRAM
I (390) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (397) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (403) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (410) spi_flash: detected chip: generic
I (414) spi_flash: flash io: qio
W (418) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (431) sleep: Configure to isolate all GPIO pins in sleep state
I (438) sleep: Enable automatic switching of GPIO sleep configuration
I (445) esp_core_dump_flash: Init core dump to flash
I (451) esp_core_dump_flash: Found partition 'coredump' @ 3c0000 65536 bytes
I (465) esp_core_dump_flash: Core dump data checksum is correct
I (465) esp_core_dump_flash: Found core dump 9156 bytes in flash @ 0x3c0000
I (472) coexist: coexist rom version e7ae62f
I (478) cpu_start: Starting scheduler on PRO CPU.
W (490) esp_qcloud_log_flash: <ESP_ERR_NVS_NOT_FOUND> NVS open for log_status log_info log_info failed
I (676) esp_qcloud_log_flash: LOG flash initialized successfully
I (676) esp_qcloud_log_flash: Log save partition subtype: label: log_info, addr:0x3d0000, offset: 0, size: 65536
I (683) esp_qcloud_log: log initialized successfully
E (688) app_main: The device has been restarted abnormally
E (695) esp_qcloud_device: Please check if the authentication information of the device is configured
E (704) esp_qcloud_device: Obtain authentication configuration information from login qcloud iothut: 
E (714) esp_qcloud_device: https://console.cloud.tencent.com/iotexplorer
E (721) esp_qcloud_device: product_id: PRODUCT_ID
E (727) esp_qcloud_device: device_name: DEVICE_NAME
E (732) esp_qcloud_device: device_secret: DEVICE_SECRET
```

Szeroy, 2022-08-17:
将freertos设置为只运行在一个内核上，程序可以跑起来了，但这样是不是另外一个内核的性能浪费了？

shixinke-orion, 2022-08-17:
双核的情况下可以做下述调整
```
(Top) → Component config → ESP System Settings → Main task core affinity
( ) CPU0
(X) CPU1
( ) No affinity
```

Szeroy, 2022-08-18:
我在程序里面设置了softAP配网模式，但是无法配网，报错了几个警告
`I (27) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (27) boot: compile time 11:56:12
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 2MB
I (50) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 0000d000 00006000
I (74) boot:  1 phy_init         RF data          01 01 00013000 00001000
I (81) boot:  2 factory          factory app      00 00 00020000 00100000
I (89) boot: End of partition table
I (93) boot_comm: chip revision: 1, min. application chip revision: 0
I (100) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=2958ch (169356) map
I (170) esp_image: segment 1: paddr=000495b4 vaddr=3ffb0000 size=03880h ( 14464) load
I (176) esp_image: segment 2: paddr=0004ce3c vaddr=40080000 size=031dch ( 12764) load
I (182) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=ae644h (714308) map
I (443) esp_image: segment 4: paddr=000fe66c vaddr=400831dc size=11458h ( 70744) load
I (472) esp_image: segment 5: paddr=0010facc vaddr=50000000 size=00010h (    16) load
I (482) boot: Loaded app from partition at offset 0x20000
I (483) boot: Disabling RNG early entropy source...
I (495) cpu_start: Pro cpu up.
I (495) cpu_start: Starting app cpu, entry point is 0x400811a4
0x400811a4: call_start_cpu1 at D:/esp-idf/esp-idf-4.4v/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (511) cpu_start: Pro cpu start user code
I (511) cpu_start: cpu freq: 160000000
I (511) cpu_start: Application information:
I (516) cpu_start: Project name:     i80_controller
I (521) cpu_start: App version:      229152f-dirty
I (527) cpu_start: Compile time:     Aug 18 2022 11:55:55
I (533) cpu_start: ELF file SHA256:  4c47628361412175...
I (539) cpu_start: ESP-IDF:          v4.4.2
I (544) heap_init: Initializing. RAM available for dynamic allocation:
I (551) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (557) heap_init: At 3FFB7848 len 000287B8 (161 KiB): DRAM
I (563) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (569) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (576) heap_init: At 40094634 len 0000B9CC (46 KiB): IRAM
I (583) spi_flash: detected chip: generic
I (587) spi_flash: flash io: dio
W (591) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (605) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
================== APPINFO ===================
    ESP_IDF VERSION: v4.4.2
    APP VERSION: 4.4v
    Compile time: Aug 18 2022 12:09:53
================================================
W (70) esp_qcloud_log_flash: <ESP_ERR_NOT_FOUND> NVS Flash init failed
W (771) esp_qcloud_log_flash: <ESP_ERR_NOT_FOUND> log_info_storage_init
I (781) esp_qcloud_log: log initialized successfully
I (781) esp_qcloud_device: --->device_name=58BF25176964
I (791) esp_qcloud_device: --->product_id=4E9T7RQ9SB
I (791) esp_qcloud_device: --->22222222222
I (801) app_main: ------>
I (811) wifi:wifi driver task: 3ffc47e0, prio:23, stack:6656, core=0
I (811) system_api: Base MAC address is not set
I (811) system_api: read default base MAC address from EFUSE
I (821) wifi:wifi firmware version: eeaa27d
I (821) wifi:wifi certification version: v7.0
I (831) wifi:config NVS flash: enabled
I (831) wifi:config nano formating: disabled
I (831) wifi:Init data frame dynamic rx buffer num: 32
I (841) wifi:Init management frame dynamic rx buffer num: 32
I (841) wifi:Init management short buffer num: 32
[... 33 lines cut ...]
   ▀▄█▀█▀▀█ ▄ ▀█▀▀▄▀▀▄█ ▄▄▀▀█  ▀▀ ▀▀█▄▀
  ▄▄█▄  ▀▄▄█▄█▀▀▀▀▄ ▄ ▄██▄▄  ████▄ ▄▀▀█
  ▄ ▄ ██▀▀▀█▀▀▄▀▀▀█ ▄▄▄▄█▄▀█▀ ▄▀▀▄▀▀█▄▀
  ▄ ▄▄▀▀▀▄ █▄ ▄ ▄▀▄  ▄█▄▄ ▄ ▀▀█▀ █  ▀▀█
  ▀▄ █▄▀▀█ ██ ▀█▀▀ █▀█▄ ▀█▀█▀ ▄▀▀█▀███▀
  ▄▄▀█ ▄▀ ▀ ██▀ █▀  ▀▄ ▄█▄▀▀█▀▀▄▀██  ▀█
  ▄  ██ ▀████▀█▀▀█▄  ▄█▄▀▄ ▀█▀▄▀▀▄▀▀██▀
  █ ▀  ▄▀ ▄▀█ ▀ █▀▀ ▄  ▄▀  ▀ ▀▄██▄▄ ▀▀█
  ▀   ▀ ▀▀▄▀█▄ █▄██▀▀▄█▄█▄▀▀▀▄█▀▀▀█ ▀
  █▀▀▀▀▀█ ▀█▀▄▄ ██▀▀ ▄▄█▀ ▀ █▄█ ▀ ██ ▀▀
  █ ███ █ █ █▄ ▀█ ▀  ▄ ▄▀▄▀▀██▀██▀█▀█ ▀
  █ ▀▀▀ █ █▀▄   ▀█▀▀█  █▄▄ ▀  ▄▄ █▄▀▀▀█
  ▀▀▀▀▀▀▀ ▀▀  ▀▀▀▀▀   ▀ ▀▀▀▀       ▀▀▀▀

I (1331) esp_qcloud_prov: If QR code is not visible, copy paste the below URL in a browser.
https://rainmaker.espressif.com/qrcode.html?data=https://iot.cloud.tencent.com/iotexplorer/device?page=softap%26productId=4E9T7RQ9SB%26ver=v1%26name=test
I (7578841) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<0,0>, prof:1
I (7578841) wifi:station: ee:1c:76:57:3a:62 join, AID=1, bgn, 20
I (7578841) esp_qcloud_wifi: STA Connecting to the AP again...
I (7578851) esp_qcloud_prov: STA Connecting to the AP again...
I (7578851) esp_qcloud_prov_smartconfig: smartconfig over
I (7579031) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
W (7579331) wifi:<ba-add>idx:2 (ifx:1, ee:1c:76:57:3a:62), tid:0, ssn:17, winSize:64
W (7595241) httpd_uri: httpd_uri: URI '/' not found
W (7595241) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI does not exist`

shixinke-orion, 2022-08-19:
手机连接 softap 前请确保 `udp_server_task` 已经运行，请检查时间戳为 `2031`  的 log 是否存在。正常 log 如下：
```
I (0) cpu_start: App cpu up.
I (345) cpu_start: Pro cpu start user code
I (345) cpu_start: cpu freq: 160000000
I (346) cpu_start: Application information:
I (348) cpu_start: Project name:     led_light
I (353) cpu_start: App version:      43e218d-dirty
I (359) cpu_start: Compile time:     Aug 19 2022 11:37:08
I (365) cpu_start: ELF file SHA256:  408ac1cde0af3437...
I (371) cpu_start: ESP-IDF:          v4.4.2
I (376) heap_init: Initializing. RAM available for dynamic allocation:
I (383) heap_init: At 3FC9FF90 len 00040070 (256 KiB): D/IRAM
I (389) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (396) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (402) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (409) spi_flash: detected chip: generic
I (413) spi_flash: flash io: qio
W (417) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (431) sleep: Configure to isolate all GPIO pins in sleep state
I (437) sleep: Enable automatic switching of GPIO sleep configuration
I (444) esp_core_dump_flash: Init core dump to flash
I (450) esp_core_dump_flash: Found partition 'coredump' @ 3c0000 65536 bytes
I (481) esp_core_dump_flash: Core dump data checksum is correct
I (481) esp_core_dump_flash: Found core dump 19588 bytes in flash @ 0x3c0000
I (484) coexist: coexist rom version e7ae62f
I (490) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
W (615) esp_qcloud_log_flash: <ESP_ERR_NVS_NOT_FOUND> NVS open for log_status log_info log_info failed
I (807) esp_qcloud_log_flash: LOG flash initialized successfully
I (807) esp_qcloud_log_flash: Log save partition subtype: label: log_info, addr:0x3d0000, offset: 0, size: 65536
I (814) esp_qcloud_log: log initialized successfully
E (819) app_main: The device has been restarted abnormally
I (827) pp: pp rom version: e7ae62f
I (830) net80211: net80211 rom version: e7ae62f
I (836) wifi:wifi driver task: 3fce7388, prio:23, stack:6656, core=0
I (841) system_api: Base MAC address is not set
I (846) system_api: read default base MAC address from EFUSE
I (872) wifi:wifi firmware version: eeaa27d
I (873) wifi:wifi certification version: v7.0
I (874) wifi:config NVS flash: enabled
I (874) wifi:config nano formating: disabled
I (878) wifi:Init data frame dynamic rx buffer num: 20
I (882) wifi:Init management frame dynamic rx buffer num: 20
I (888) wifi:Init management short buffer num: 32
I (892) wifi:Init dynamic tx buffer num: 20
I (896) wifi:Init static tx FG buffer num: 2
I (900) wifi:Init static rx buffer size: 1600
I (904) wifi:Init static rx buffer num: 6
I (908) wifi:Init dynamic rx buffer num: 20
I (912) wifi_init: rx ba win: 6
I (916) wifi_init: tcpip mbox: 20
I (920) wifi_init: udp mbox: 20
I (923) wifi_init: tcp mbox: 20
I (927) wifi_init: tcp tx win: 24000
I (931) wifi_init: tcp rx win: 24000
I (936) wifi_init: tcp mss: 1440
I (940) wifi_init: WiFi IRAM OP enabled
I (944) wifi_init: WiFi RX IRAM OP enabled
I (971) phy_init: phy_version 503,13653eb,Jun  1 2022,17:47:08
I (1010) wifi:mode : softAP (7c:df:a1:e0:41:a9)
I (1012) wifi:Total power save buffer number: 10
I (1012) wifi:Init max length of beacon: 752/752
I (1013) wifi:Init max length of beacon: 752/752
I (1018) esp_qcloud_prov_softapconfig: Starting provisioning
I (1024) wifi:mode : sta (7c:df:a1:e0:41:a8)
I (1028) wifi:enable tsf
I (1034) wifi:mode : sta (7c:df:a1:e0:41:a8) + softAP (7c:df:a1:e0:41:a9)
I (1040) wifi:Total power save buffer number: 10
I (2014) wifi:Total power save buffer number: 10
[... 58 lines cut ...]
I (273732) wifi:state: assoc -> run (10)
I (275895) wifi:connected with Xiaomi_32BD, aid = 1, channel 11, BW20, bssid = 88:c3:97:9e:32:be
I (275896) wifi:security: WPA2-PSK, phy: bgn, rssi: -29
I (275906) wifi:pm start, type: 1

I (275907) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 14, mt_pti: 25000, mt_time: 10000
W (275961) wifi:<ba-add>idx:0 (ifx:0, 88:c3:97:9e:32:be), tid:0, ssn:5, winSize:64
I (276037) wifi:BcnInt:102400, DTIM:1
I (276827) esp_netif_handlers: sta ip: 192.168.31.43, mask: 255.255.255.0, gw: 192.168.31.1
I (276828) esp_qcloud_wifi: Connected with IP Address:192.168.31.43
I (276834) esp_qcloud_prov: Connected with IP Address:192.168.31.43
I (276840) esp_qcloud_prov: sendto, data: {"cmdType":2,"productId":"J7LC0IF6CL","deviceName":"t1","protoVersion":"2.0"}
I (276852) esp_qcloud_prov: Shutting down socket
I (277961) wifi:station: 06:8c:09:43:fa:19 leave, AID = 1, bss_flags is 658531, bss:0x3fcf7c2c
I (277963) wifi:new:<11,0>, old:<11,2>, ap:<11,2>, sta:<11,0>, prof:1
W (277966) wifi:<ba-del>idx
W (277968) wifi:<ba-del>idx
I (277971) wifi:mode : sta (7c:df:a1:e0:41:a8)
I (277975) wifi_prov_mgr: Provisioning stopped
I (277984) esp_qcloud_timesync: Initializing SNTP. Using the SNTP server: pool.ntp.org
I (277990) esp_qcloud_mqtt: Initialising MQTT
I (277995) esp_qcloud_mqtt: Connecting to mqtt://J7LC0IF6CL.iotcloud.tencentdevices.com:1883
I (278003) esp_qcloud_mqtt: Waiting for MQTT connection. This may take time.
I (278116) esp_qcloud_mqtt: MQTT Connected
```

Szeroy, 2022-08-22:
问题已经解决了，谢谢