espressif/esp-qcloud #33: ESP32C3 配网蓝牙连接出错

meiliang1123, 2022-11-21:
```
I (1945) esp_qcloud_prov: If QR code is not visible, copy paste the below URL in a browser.
https://rainmaker.espressif.com/qrcode.html?data=https://iot.cloud.tencent.com/iotexplorer/device?page=softap%26productId=0IPCY9TC3G%26ver=v1%26name=tcloud_XXX
I (1968) wifi:mode : sta (f4:12:fa:18:b4:58)
I (2022) smartconfig: SC version: V3.0.1
W (2023) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK
I (2023) BTDM_INIT: BT controller compile version [4ec6fee]
I (2029) coexist: coexist rom version 9387209
I (2034) BTDM_INIT: Bluetooth MAC: f4:12:fa:18:b4:5a

I (2051) esp_qcloud_prov_ble: BD ADDR: f4:12:fa:18:b4:5a
I (2053) esp_qcloud_prov_ble: BLUFI init finish
I (8635) wifi:ic_enable_sniffer
I (8637) smartconfig: Start to find channel...
I (8637) esp_qcloud_prov_smartconfig: Scan done
W (42883) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x13
**E (42885) BT_HCI: CC evt: op=0x2022, status=0x2**
I (42885) esp_qcloud_prov_ble: BLUFI ble connect
I (42889) esp_qcloud_prov_ble: BLUFI ble disconnect
W (44785) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x13
```
配置并启动成功，扫码后蓝牙可发现，点击蓝牙设备后超时提示蓝牙连接失败，log报错如上， `BT_HCI: CC evt: op=0x2022, status=0x2` 是什么原因？如何解决？

shixinke-orion, 2022-11-22:
不同 IDF 版本，不同项目使用的蓝牙配置不一致，有些配置会存放在 nvs 中，在配置发生变化时某些手机安全特性会阻止连接，此问题常出现在开发阶段。解决方案：清除配对，并擦除 flash  重新烧录。

amibk, 2023-02-09:
hello, I have a same problem, this is IDF log, and  , has your problem been solved??

I (567) pm: Frequency switching config: CPU_MAX: 160, APB_MAX: 80, APB_MIN: 80, Light sleep: ENABLED
I (568) sleep: Enable automatic switching of GPIO sleep configuration
D (569) esp_qcloud_storage: <ESP_ERR_NVS_NOT_FOUND> Get value for given key, key: wifi_config
I (569) main: qcloud net get wifi config info
W (585) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK
I (586) BTDM_INIT: BT controller compile version [421c279]
I (586) phy_init: phy_version 912,d001756,Jun  2 2022,16:28:07
W (587) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (623) BTDM_INIT: Bluetooth MAC: a0:76:4e:70:bf:86

I (667) esp_qcloud_prov_ble: BD ADDR: a0:76:4e:70:bf:86
I (669) esp_qcloud_prov_ble: BLUFI init finish
W (670) BT_BTM: BTM_BleWriteAdvData, Partial data write into ADV
W (35832) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (41804) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (43132) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (47471) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (57598) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (78937) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (81260) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (116186) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (118841) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e

user468565, 2023-11-05:
请问一下这个问题解决了吗，解决的办法是什么，我的设备也出现的这个问题，现在也不知道该怎么解决这个问题。