espressif/esp-qcloud #34: 在ESP32C3中无法进行蓝牙辅助配网

amibk, 2023-02-10:
这里是ESP-IDF输出的日志内容：
`
I (567) pm: Frequency switching config: CPU_MAX: 160, APB_MAX: 80, APB_MIN: 80, Light sleep: ENABLED
I (568) sleep: Enable automatic switching of GPIO sleep configuration
D (569) esp_qcloud_storage: <ESP_ERR_NVS_NOT_FOUND> Get value for given key, key: wifi_config
I (569) main: qcloud net get wifi config info
W (585) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK
I (586) BTDM_INIT: BT controller compile version [421c279]
I (586) phy_init: phy_version 912,d001756,Jun 2 2022,16:28:07
W (587) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (623) BTDM_INIT: Bluetooth MAC: a0:76:4e:70:bf:86

I (667) esp_qcloud_prov_ble: BD ADDR: a0:76:4e:70:bf:86
I (669) esp_qcloud_prov_ble: BLUFI init finish
W (670) BT_BTM: BTM_BleWriteAdvData, Partial data write into ADV
W (35832) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (41804) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (43132) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (47471) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (57598) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (78937) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (81260) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (116186) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (118841) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
`
发起配网的客户端为腾讯连连小程序，可以正常发现设备，但是无法进行连接，在连接时会输出这样的日志。
尝试过的ESP-IDF版本有v4.3.2、v4.3.4、v4.4.4都会有同样的问题
对了，在每次更换版本重新编译固件时，都会执行一次idf.py erase_flash清除数据再进行烧录，用于保证每次烧录时FLASH为纯净状态

感谢您的任何回答。

shixinke-orion, 2023-02-10:
是否有清除手机配对？ 如清除后仍然无法连接，能否测试 IDF 中的 blufi 示例，看看能否正常运行.

shixinke-orion, 2023-02-17:
是否有验证过不同版本的 IDF？

amibk, 2023-02-19:
> 对 blufi 示例是否有验证过不同 release 版本的 IDF？每个 IDF 版本切换时需要全擦除 flash。
> 
> 排查思路：
> 
>     * 可以尝试 wifi 是否能正常连接，用于排查 RF 问题
> 
>     * 检查板载天线 0 欧姆电阻是否正常焊接
> 
>     * 更换蓝牙协议栈，在 menuconfig 中切换 nimble 与 bluedroid
> 
>     * 启用全校准，在 menuconfig 中 `Component config → PHY -> [ ] Store phy calibration data in NVS` 关闭 phy 校准数据存储
> 
>     * 更换测试用开发板、手机、环境，进行交叉验证

WIFI可以正常连接，更换过4.3.2、4.3.4、4.4.4版本的IDF进行尝试，结果相同，
由于一直采用iPhone手机进行操作，后来根据Apple发布的[https://developer.apple.com/library/archive/qa/qa1931/_index.html]猜测可能是iPhone手机的限制，所以在回调中加入了符合iPhone规则的参数进行更新

修改文件： esp-qcloud/src/provisioning/esp_qcloud_prov_ble.c 
第181行加入：
`esp_ble_conn_update_params_t conn_params = {0};
        memcpy(conn_params.bda, param->connect.remote_bda, sizeof(esp_bd_addr_t));
        /* For the IOS system, please reference the apple official documents about the ble connection parameters restrictions. */
        conn_params.latency = 0;
        conn_params.max_int = 0x20;    // max_int = 0x20*1.25ms = 40ms
        conn_params.min_int = 0x10;    // min_int = 0x10*1.25ms = 20ms
        conn_params.timeout = 400;    // timeout = 400*10ms = 4000ms
        BLUFI_INFO("ESP_GATTS_CONNECT_EVT, conn_id %d, remote %02x:%02x:%02x:%02x:%02x:%02x:",
                 param->connect.conn_id,
                 param->connect.remote_bda[0], param->connect.remote_bda[1], param->connect.remote_bda[2],
                 param->connect.remote_bda[3], param->connect.remote_bda[4], param->connect.remote_bda[5]);
        //start sent the update connection parameters to the peer device.
        esp_ble_gap_update_conn_params(&conn_params);`

但是实际执行情况如下，无法进行更新

`I (34626) BT_BLUFI: 
device is connected 49:14:27:81:19:9f, server_if=3,reason=0x0,connect_id=3

W (34627) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x8
I (34627) BT_BLUFI: 
device is disconnected 49:14:27:81:19:9f, server_if=3,reason=0x8,connect_id=3

E (34629) BT_HCI: CC evt: op=0x2022, status=0x2
W (34629) esp_qcloud_prov_ble: ESP_BLUFI_EVENT_BLE_CONNECT
I (34630) BLUFI_EXAMPLE: ESP_GATTS_CONNECT_EVT, conn_id 0, remote 49:14:27:81:19:9f:
I (34631) esp_qcloud_prov_ble: BLUFI ble disconnect
W (34632) BT_L2CAP: L2CA_UpdateBleConnParams - unknown BD_ADDR 49142781199f
E (34632) BT_APPL: Update connection parameters failed!`
继续尝试解决中

shixinke-orion, 2023-02-20:
方便提供下 ios 版本及测试手机型号吗？

amibk, 2023-02-20:
> 方便提供下 ios 版本及测试手机型号吗？

iOS version: iOS 16.3(20D47)
iPhone Model: iPhone13

shixinke-orion, 2023-02-22:
我使用 iphone14 单独跑 blufi 示例是正常的。使用 EspBlufi App 可以正常连接通信。你可以试试看
IDF commit: 603fc32e24073db70d8ffcca669739e27f73f942

amibk, 2023-03-23:
> 我使用 iphone14 单独跑 blufi 示例是正常的。使用 EspBlufi App 可以正常连接通信。你可以试试看 IDF commit: 603fc32e24073db70d8ffcca669739e27f73f942

检出到603fc32e24073db70d8ffcca669739e27f73f942进行尝试，iPhone13使用EspBlufi App有概率连接上保持通讯，同样出现了连接成功立即断开的情况，Android使用EspBlufi App一直正常，其中发现Andrioid系统的手机设备一直都是正常，即使使用其他不同版本的IDF，Android也一直正常未出现问题，注意到603fc32e24073db70d8ffcca669739e27f73f942属于IDF5.X分支，回到esp-qcloud项目本身，由于并未适配IDF5.X，切回IDF4.X分支后，目前是如果用户使用iOS系统设备配网，则会采用AP模式配网，使用Android设备配网，正常使用Blufi配网，暂时以此方案处理。
产品已经量产上市场了，后续发现能完美解决iOS系统中Blufi配网问题再来分享处理办法

closed 2023-03-23 by amibk

shixinke-orion, 2023-04-20:
可以尝试ble_gap_set_prefered_default_le_phy(BLE_GAP_LE_PHY_1M_MASK, BLE_GAP_LE_PHY_1M_MASK);

jiafeng-1, 2023-08-17:
@shixinke-orion 
我遇到了相同的问题，Android配网没有问题，但是IOS设备不可以。尝试了大概两三台iPhone手机都是不可以。Android都可以
使用的手机型号：iPhone 8 
系统版本：16.6

另一台手机型号：iPhone 12

系统版本：16.2

EspBlufi 软件版本：1.3.1

ESP32 idf版本： ESP-IDF v4.4.4-dirty
使用的官方提供的Blufi例程，没有修改menuconfig 
```
iPhtone8 --- IOS 配网过程日志
I (2750) BLUFI_EXAMPLE: BLUFI BLE is not connected yet

D (15150) BT_HCI: Receive packet event_code=0x3e

D (15150) BT_HCI: BLE HCI(id=62) event = 0x0a)
D (15150) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (15160) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (15160) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (15170) BT_HCI: btu_free_quick_timer Unable to find expected alarm in hashmap
D (15180) BT_HCI: btu_free_quick_timer Unable to find expected alarm in hashmap
D (15190) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (15190) BT_HCI: HCI Enqueue Comamnd opcode=0x41d

D (15200) BT_HCI: Receive packet event_code=0xf

D (15200) BT_HCI: Receive packet event_code=0xc

D (15210) BT_HCI: HCI Enqueue Comamnd opcode=0x2016

D (15210) BT_HCI: Receive packet event_code=0xf

D (15250) BT_HCI: Receive packet event_code=0x3e

D (15250) BT_HCI: BLE HCI(id=62) event = 0x04)
D (15250) BT_HCI: HCI Enqueue Comamnd opcode=0x2022

D (15250) BT_HCI: Receive packet event_code=0xe

D (15260) BT_HCI: btu_hcif_command_complete_evt

D (15260) BT_BLUFI: blufi profile cb event = e

I (15270) BT_BLUFI: 
device is connected 51:60:17:58:bf:00, server_if=3,reason=0x0,connect_id=3

I (15280) BLUFI_EXAMPLE: BLUFI ble connect

D (15280) BT_HCI: HCI Enqueue Comamnd opcode=0x200a

D (15290) BT_HCI: Receive packet event_code=0xe

D (15290) BT_HCI: btu_hcif_command_complete_evt

D (15340) BT_HCI: Receive packet event_code=0x3e

D (15340) BT_HCI: BLE HCI(id=62) event = 0x07)
D (17050) BT_HCI: Receive packet event_code=0x5

W (17050) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x8
D (17050) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (17060) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (17060) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (17070) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (17080) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (17090) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
[... 228 lines cut ...]
D (17130) BT_HCI: HCI Enqueue Comamnd opcode=0x200a

D (17130) BT_HCI: Receive packet event_code=0xe

D (17140) BT_HCI: btu_hcif_command_complete_evt

D (17140) BT_HCI: HCI Enqueue Comamnd opcode=0x2006

D (17150) BT_HCI: Receive packet event_code=0xe

D (17150) BT_HCI: btu_hcif_command_complete_evt

D (17150) BT_HCI: HCI Enqueue Comamnd opcode=0x200a

D (17160) BT_HCI: Receive packet event_code=0xe

D (17160) BT_HCI: btu_hcif_command_complete_evt

D (17170) BT_HCI: HCI Enqueue Comamnd opcode=0x200a

D (17180) BT_HCI: Receive packet event_code=0xe

D (17180) BT_HCI: btu_hcif_command_complete_evt

```

shixinke-orion, 2023-08-21:
> > 可以尝试在 blufi 代码初始化后增加下述测试代码： 对于 nimble 协议栈： `ble_gap_set_prefered_default_le_phy(BLE_GAP_LE_PHY_1M_MASK, BLE_GAP_LE_PHY_1M_MASK)` 对于 bluedroid 协议栈： `esp_ble_gap_set_prefered_default_phy(ESP_BLE_GAP_PHY_1M_PREF_MASK, ESP_BLE_GAP_PHY_1M_PREF_MASK)`
> 
> 加入这个好像需要支持BLE5.0才可以，目前menuconfig 默认是没有开启的，我开启后，依旧无法配网。但是大概在8月11日，还有同事，使用苹果手机配网成功，8月15日就不可以用了，今天尝试官方demo发现也不行。麻烦看看是啥问题。

这些选项只是为了配置通信速率为 1M，与 BLE 5.0 没有关系，我们之前遇到过 ios 手机 phy 速率更新事件不匹配导致的连接失败，如果是这个问题，这些改动是有效的。同时你可升级到 v4.4.5 测试看看。