espressif/esp-qcloud #36: ESP-IDF5.0.1版本支持吗？我下载后编译不通过

king592, 2023-04-01:
我将文件下载进来，然后放入IDF的components目录并且改名为esp-qcloud，但是编译报错，是版本不支持吗

shixinke-orion, 2023-04-04:
现已支持，请拉取最新提交

king592, 2023-05-30:
你好我填入三元组之后，使用ESP32编译是可以通过的，但是芯片更换成ESP32-C3或者S3编译就不行了

king592, 2023-05-30:
@shixinke-orion 报错信息如下

CMake Error at D:/Install/ESP-IDF/esp-idf/tools/cmake/targets.cmake:19 (message):
  IDF_TARGET in CMake cache does not match IDF_TARGET environment variable.
  To change the target, clear the build directory and sdkconfig file, and
  build the project again
Call Stack (most recent call first):
  D:/Install/ESP-IDF/esp-idf/tools/cmake/project.cmake:7 (__target_init)
  CMakeLists.txt:33 (include)

-- Configuring incomplete, errors occurred!
See also "E:/demo/led_light/led_light/build/CMakeFiles/CMakeOutput.log".

shixinke-orion, 2023-05-31:
可能是你的 IDF 没有包含 ESP32C3 工具链信息

king592, 2023-05-31:
但是我的IDF编译其他的ESP32-C3工程是正常的，blink和helloword都是可以编译的 @shixinke-orion

shixinke-orion, 2023-05-31:
可以把 https://github.com/espressif/esp-qcloud/blob/master/src/iothub/esp_qcloud_ota.c#L278 改成 `if 1` 试一下