espressif/esp-qcloud #37: 进行OTA升级发现升级失败

Kingwulin, 2023-05-31:
大概半年前还可以升级的，最近一直升级不了，然后重新拉取了代码，编译led light 例子，发现也是升级失败，附件是升级失败的log，麻烦帮忙分析下，谢谢
[esp-qcloud升级失败.txt](https://github.com/espressif/esp-qcloud/files/11609189/esp-qcloud.txt)

shixinke-orion, 2023-05-31:
提供下编译生成的 sdkconfig 看看

Kingwulin, 2023-05-31:
[sdkconfig.txt](https://github.com/espressif/esp-qcloud/files/11610289/sdkconfig.txt)

shixinke-orion, 2023-05-31:
可以把 https://github.com/espressif/esp-qcloud/blob/master/src/iothub/esp_qcloud_ota.c#L278 改成 if 1 试一下

Kingwulin, 2023-06-01:
感谢，可以了，需要打开使用https的宏控

Kingwulin, 2023-06-12:
[image]
你好，发现提示升级成功了，但是实际上版本没有变化，打印的还是版本v0.0.1
我修改的是这里的版本 https://github.com/espressif/esp-qcloud/blob/7091584d9950d5a0d2af90cf66280e2b7d2ec67b/examples/led_light/main/app_main.c#LL285C21-L285C53

Kingwulin, 2023-06-12:
固件如果是直接手动烧录的版本号就会变成0.0.2，ota的一直是0.0.1，请教下这是怎么回事

shixinke-orion, 2023-06-13:
检查下固件类型 (MCU/模组)，信息有没有上传错。创建 ota 任务时版本号一定要与固件实际版本号一致。

Kingwulin, 2023-06-13:
>实际固件是模块的，但是发现最近新的设备都上报成了mcu，这个模块和mcu代码上怎么区分

shixinke-orion, 2023-06-13:
`mcu固件版本` 这个字段 应该就是由 https://github.com/espressif/esp-qcloud/blob/7091584d9950d5a0d2af90cf66280e2b7d2ec67b/examples/led_light/main/app_main.c#LL285C21-L285C53 上报的。https://cloud.tencent.com/document/product/1081/34916 中的 `设备基础信息上报` 章节有相关描述。

Kingwulin, 2023-06-13:
代码上是否可以修改成上报模块版本的

shixinke-orion, 2023-06-13:
`esp_qcloud_device_add_fw_version` 就是更改版本的

Kingwulin, 2023-06-13:
那我前面这种平台提示升级成功，实际没有升级成功有办法解决吗

shixinke-orion, 2023-06-13:
提供设备完整的 log 看下为什么没升级成功

shixinke-orion, 2023-06-13:
检查下分区表是不是被改动了，是不是缺少了 otadata 分区。

Kingwulin, 2023-06-13:
[image]

Kingwulin, 2023-06-13:
[image]

Kingwulin, 2023-06-13:
[image]
添加这个就可以了，感谢大佬