espressif/esp-qcloud #38: 无法启用APP_CPU

SteamedCharger, 2023-09-04:
ESP-IDF版本：v4.4.5
模块：ESP-WROOM-32

将"Run FreeRTOS only on first core"设置为false之后，系统无法正常启动：
[image]

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:5956
load:0x40078000,len:15584
ho 0 tail 12 room 4
load:0x40080400,len:4452
0x40080400: _init at ??:?

entry 0x400806a0
I (29) boot: ESP-IDF v4.4.5 2nd stage bootloader
I (29) boot: compile time 18:31:50
I (29) boot: chip revision: v3.1
I (32) qio_mode: Enabling default flash chip QIO
I (37) boot.esp32: SPI Speed      : 80MHz
I (42) boot.esp32: SPI Mode       : QIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 0000d000 00008000
I (75) boot:  1 fctry            WiFi data        01 02 00015000 00004000
I (82) boot:  2 log_status       WiFi data        01 02 00019000 00004000
I (90) boot:  3 otadata          OTA data         01 00 0001d000 00002000
I (97) boot:  4 phy_init         RF data          01 01 0001f000 00001000
I (105) boot:  5 ota_0            OTA app          00 10 00020000 001d0000
I (112) boot:  6 ota_1            OTA app          00 11 001f0000 001d0000
I (120) boot:  7 coredump         Unknown data     01 03 003c0000 00010000
I (128) boot:  8 log_info         Unknown data     01 fe 003d0000 00010000
I (135) boot:  9 reserved         Unknown data     01 ff 003e0000 00020000
I (143) boot: End of partition table
I (147) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=35da4h (220580) map
I (216) esp_image: segment 1: paddr=00055dcc vaddr=3ffbdb60 size=0661ch ( 26140) load
I (224) esp_image: segment 2: paddr=0005c3f0 vaddr=40080000 size=03c28h ( 15400) load
I (230) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=f66e4h (1009380) map
I (509) esp_image: segment 4: paddr=0015670c vaddr=40083c28 size=1aa0ch (109068) load
I (559) boot: Loaded app from partition at offset 0x20000
I (559) boot: Disabling RNG early entropy source...
I (571) cpu_start: Pro cpu up.
I (571) cpu_start: Starting app cpu, entry point is 0x40080e20
0x40080e20: call_start_cpu1 at D:/Users/tonny/esp/esp-idf/components/esp_system/port/cpu_start.c:147

I (0) cpu_start: App cpu up.
I (587) cpu_start: Pro cpu start user code
I (587) cpu_start: cpu freq: 240000000
I (587) cpu_start: Application information:
I (592) cpu_start: Project name:     led_light
I (597) cpu_start: App version:      1
I (602) cpu_start: Compile time:     Sep  4 2023 18:31:28
I (608) cpu_start: ELF file SHA256:  e360462e1f920c35...
I (614) cpu_start: ESP-IDF:          v4.4.5
I (619) cpu_start: Min chip rev:     v0.0
I (623) cpu_start: Max chip rev:     v3.99
I (628) cpu_start: Chip rev:         v3.1
I (633) heap_init: Initializing. RAM available for dynamic allocation:
I (640) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (646) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (652) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (658) heap_init: At 3FFCCAB8 len 00013548 (77 KiB): DRAM
I (664) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (671) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (677) heap_init: At 4009E634 len 000019CC (6 KiB): IRAM
I (684) spi_flash: detected chip: generic
I (688) spi_flash: flash io: qio
I (692) esp_core_dump_flash: Init core dump to flash
I (698) esp_core_dump_flash: Found partition 'coredump' @ 3c0000 65536 bytes
I (711) esp_core_dump_flash: Core dump data checksum is correct
I (712) esp_core_dump_flash: Found core dump 7844 bytes in flash @ 0x3c0000
I (720) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

``` 
然后串口不再输出任何数据。

SteamedCharger, 2023-09-04:
换了几个版本的idf，发现5.0.3也有相同的问题；5.1.1可以正常启动，但无法配网（可以获取到SSID和PASSWD但无法连接，没有错误信息）

SteamedCharger, 2023-09-05:
问题解决了。5.1.1在esp_wifi_restore();后默认为softAP模式，需要切换到STA才能连网。

shixinke-orion, 2023-09-06:
是个线程调度的问题，可以临时把[此处](https://github.com/espressif/esp-qcloud/blob/master/src/utils/esp_qcloud_reboot.c#L42 )更改为：
```
#define CONFIG_UNBROKEN_RECORD_TASK_DEFAULT_PRIOTY (ESP_TASK_MAIN_PRIO - 1)
```

对于无法配网，请提供详细的 log

SteamedCharger, 2023-09-06:
十分感谢。

配网问题已解决：
```c
// app_main.c
static esp_err_t get_wifi_config(wifi_config_t *wifi_cfg, uint32_t wait_ms)
{
    /* 省略 */

    /**< Reset wifi and restart wifi */
    esp_wifi_restore();
    esp_wifi_set_mode(WIFI_MODE_STA); //加入这一行
    esp_wifi_start();

    /* 省略 */
}
```

closed 2023-09-10 by SteamedCharger